Ankermgmt/ankermake-m5-protocol #11: How to get Serial Number and Mac Address of Printer

spuder, 2023-03-23:
The following commands require knowing the mac address and serial number of the printer

```
./ankerctl.py http calc-check-code
```
```
./ankerctl.py http  calc-sec-code
```

Figure out how to get those values and document it
If possible add automation to simplify fetching those values for the user
Assume users will have 0 to many printers. 

### Research

Theory 1: Is it possible to get this from the login.json? 
Result: Maybe? 

Running this command `./ankerctl.py config decode` will parse the login.json. On my machine (osx), there is a `    'mac_addr': '',` key/value, but the value is blank.

chrivers, 2023-03-23:
Yes, it's in fact already imported to your local config by `config import`.

Try `config show` to see some highlights of your current config.

On Linux, the path is `~/.config/ankerctl/default.json`

There you can see the details collected. It's:

  - 1 account
  - any number of printers (for that account)

The reason the file is called `default.json`, is because I want to make it possible to switch to a different "profile" (i.e., a different `<name>.json`), for use with another account.

Now, the reason that these commands seem very low-level, is, well, because they're meant to be :smiley: 

But I guess we could default to the values currently saved to local config, if not specified. Patches welcome :)

spuder, 2023-03-23:
Ah, thank you. I was looking at `config decode` , when I should have been looking at `config show`

closed 2023-03-26 by thomasjpatterson