Ankermgmt/ankermake-m5-protocol #12: json is not machine parseable

spuder, 2023-03-23:
jq doesn't like the single quotes that ` ./ankerctl.py config decode` returns

```
./ankerctl.py config decode 2>/dev/null | jq .     
parse error: Invalid numeric literal at line 2, column 14
```

```
jq empty /tmp/foo.json 
parse error: Invalid numeric literal at line 2, column 14
```

## Workaround

- Replace single quotes with double quotes
- Replace booleans like `None` with strings

```
./ankerctl.py config decode 2>/dev/null |  sed "s/'/\"/g" | sed 's/None/"None"/g' | jq -r
```

## Solution

Change the single quotes to double quotes

```
{
    'ab_code': 'US',
    'auth_token': 'REDACTED',
    'avatar': '',
```

```
{
    "ab_code": "US",
    "auth_token": "REDACTED",
    "avatar": '',
```  

## Priority

This is a low priority, most people won't need to interact with this file much.

chrivers, 2023-03-23:
Fixed in branch `chrivers/ankerctl`, which will be merged soon-ish :)

chrivers, 2023-03-23:
Branch `chrivers/ankerctl` is now merged - this is fixed :)

PR #13 (2023-03-23): Json output
Fixes #12 

```
 ./ankerctl.py config decode -v | jq
[*] Loading file..
{
  "ab_code": "US",
 }
```
closed 2023-03-23 by chrivers