Ankermgmt/ankermake-m5-protocol #17: 'libflagship' is not defined

spuder, 2023-03-25:
On this commit 92c6c67ef048bb5c75f8aad6dd61363493b1f05b

`pip3 install -r requirements.txt`

```
./ankerctl.py pppp lan-search   
Traceback (most recent call last):
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/./ankerctl.py", line 343, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/./ankerctl.py", line 174, in pppp_lan_search
    api = AnkerPPPPApi.open_broadcast(timeout=1.0)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/ppppapi.py", line 30, in open_broadcast
    addr = ("255.255.255.255", libflagship.pppapi.LAN_SEARCH_PORT)
                               ^^^^^^^^^^^
NameError: name 'libflagship' is not defined
```

chrivers, 2023-03-25:
I pushed a hotfix - can you try again?

spuder, 2023-03-25:
Fixed

Printer is turned off so this error is expected. 
```
./ankerctl.py pppp lan-search
[*] TX f1300000
[E] No printers responded within timeout. Are you connected to the same network as the printer?
```

closed 2023-03-25 by spuder

chrivers, 2023-03-25:
Thanks for testing, and the quick report :)