Ankermgmt/ankermake-m5-protocol #20: Windows can not find AppData path

spuder, 2023-03-28:
Running on windows, seeing this error

```bash
python3.11.exe .\ankerctl.py config import
Traceback (most recent call last):
  File "C:\Users\spuder\Resilio Sync\Code\github\Ankermgmt\ankermake-m5-protocol\ankerctl.py", line 343, in <module>
    main()
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\core.py", line 1654, in invoke
    super().invoke(ctx)
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\Resilio Sync\Code\github\Ankermgmt\ankermake-m5-protocol\ankerctl.py", line 48, in main
    env.config   = cli.config.configmgr()
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\Resilio Sync\Code\github\Ankermgmt\ankermake-m5-protocol\cli\config.py", line 74, in configmgr
    return AnkerConfigManager(PlatformDirs("ankerctl"), classes=(Config, Account, Printer))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\spuder\Resilio Sync\Code\github\Ankermgmt\ankermake-m5-protocol\cli\config.py", line 17, in __init__
    dirs.user_config_path.mkdir(exist_ok=True)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.752.0_x64__qbz5n2kfra8p0\Lib\pathlib.py", line 1116, in mkdir
    os.mkdir(self, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\spuder\\AppData\\Local\\ankerctl\\ankerctl'
```

For some reason config.py is unable to create the user config directory `dirs.user_config_path.mkdir(exist_ok=True)`

[User in discord reports similar problem](https://discordapp.com/channels/956636378458898542/1033511859044765807/1090107630879506524) (but not identical problem) on windows

## TODO:

- Figure out why config.py line 17 is trying to create `AppData\Local\ankerctl\ankerctl` as the config path (linux uses `/root/.config/ankerctl/default.json`)

chrivers, 2023-03-28:
I'm using the `platformdirs` package, which tries to do a "reasonable" thing on the various platforms.

I guess the problem is that we're only trying to `mkdir` the innermost dir `..\ankerctl`, not the upper dir from that (which is also missing).

I pushed an update which should fix this - can you try again?

spuder, 2023-03-28:
```
python3.11.exe .\ankerctl.py config import
[*] Loading cache..
[*] Initializing API..
[*] Requesting profile data..
[*] Requesting printer list..
[*] Requesting pppp keys..
[*] Adding printer [REDACTED]
[*] Finished import
```

closed 2023-03-28 by spuder