Ankermgmt/ankermake-m5-protocol #25: ankerctl config import broken

spuder, 2023-04-03:
Seeing this error when running an ankerctl config import (on the chrivers/pppp-improvements branch commit: 0339ef71782e6e8e4bdafd507659cfa7ad365b68)

```
./ankerctl.py config import
[*] Loading cache..
[*] Initializing API..
[*] Requesting profile data..
Traceback (most recent call last):
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/./ankerctl.py", line 369, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/./ankerctl.py", line 337, in config_import
    config = cli.config.load_config_from_api(auth_token, region, env.insecure)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/cli/config.py", line 91, in load_config_from_api
    profile = ppapi.profile()
              ^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/httpapi.py", line 13, in wrapper
    return func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/httpapi.py", line 90, in profile
    return self._get("/profile", headers={"X-Auth-Token": self._auth})
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/httpapi.py", line 33, in wrapper
    raise ValueError(f"API request failed: {data.status_code} {data.reason}")
ValueError: API request failed: 401 Unauthorized
```

## Theories

I haven't opened the anker slicer in a couple of days/weeks. Maybe my credentials expired?

spuder, 2023-04-03:
Opening the slicer shows this error, which seems to indicate that ankerctl can't handle when reauthentication is required

[image]

chrivers, 2023-04-03:
Yes, indeed.

Another issue is having to enter a CAPTCHA if you try to login too many times in a row (*even if successful*), which is also very hard to handle in a CLI tool.

I'll add some exception handling to this at some point, to give it a better presentation.

commit 49d56773 (2023-04-03): + Added exception handling to "config import". Fixes #25
closed 2023-04-03 by spuder