Ankermgmt/ankermake-m5-protocol #34: FileUploadInfo.__init__() missing 1 required positional argument: 'machine_id'

therealmacjeezy, 2023-04-05:
When attempting to print from prusaslicer, it said the transfer was complete, however the printer wasn't actually doing anything. Attempted to use the CLI to print the file (`./ankerctl.py pppp print-file <filepath>`) and noticed the error message `FileUploadInfo.__init__() missing 1 required positional argument: 'machine_id'`. 

I was able to resolve the issue by adding the md5 type and assigning the variables in the `from_file` function in `ppppapi.py`

**Before:** `return cls(data, filename, user_name, user_id, machine_id, type=0)`
**After:**  `return cls(data, filename, user_name=user_name, user_id=user_id, machine_id=machine_id, type=0, md5=hashlib.md5(data).hexdigest())`

After saving the above change, I was able to use prusaslicer to "Upload/Print" in a matter of seconds. 

**platform:** Mac Mini (m1)
**python version:** 3.10.0

chrivers, 2023-04-05:
Thank you for testing, and for the detailed error report.

It turned

therealmacjeezy, 2023-04-05:
Just pulled down the fix and was able to print through prusaslicer as expected!

Thank you :)

closed 2023-04-05 by therealmacjeezy