Ankermgmt/ankermake-m5-protocol #38: Trouble pulling image for dockerfile

yourjelly, 2023-04-06:
`Error response from daemon: pull access denied for ankerctl/ankerctl, repository does not exist or may require 'docker login': denied: requested access to the resource is denied`

spuder, 2023-04-06:
Thanks for reporting this issue. 

Since the docker container doesn't exist in any docker registry (yet), you will have to build the container with `docker build -t ankerctl .` Or if you have docker compose just run `docker compose up` and it should build it automatically. 

Could you show me the command you are running?

yourjelly, 2023-04-06:
I'm using unraid compose plugin to build.
I have now got it to build, but it is stuck in a crash loop will create a new issue