Ankermgmt/ankermake-m5-protocol #44: Add it to AnkerMake app

imadcat, 2023-04-09:
[## Upcoming and planned features
](https://github.com/Ankermgmt/ankermake-m5-protocol#upcoming-and-planned-features)
```
...
 - Integration into other software. Home Assistant? Prusa Slicer?
 ```

Wouldn't it be more logical if this tool is integrated into [the official AnkerMake App](https://github.com/ankermake/AnkerMake-Slicer)?

thomasjpatterson, 2023-04-09:
Thank your for opening a ticket. Currently, with AnkerMake switching their slicer to a Prusa backend we are holding off on any decision regarding the AnkerMake apps.

chrivers, 2023-04-09:
I mean, in a way, but mostly no :sweat_smile: 

This tool is in no way associated with Anker. This project was started, because people want to use the printer with Free and Open Source software, and be able to control it freely.

To me, it seems like Anker has somewhat different priorities. And since we're not affiliated with them in any way, we have no control over the AnkerMake Apps.

closed 2023-04-09 by chrivers