Ankermgmt/ankermake-m5-protocol #45: i installed python3 .but when i go over to : pip3 install -r requirements.txt ,then i get the message error requirements not found on ...........................ect

Jean0726, 2023-04-10:
[image]
[image]

LazeMSS, 2023-04-10:
You did not checkout the code or not in the right folder

Jean0726, 2023-04-10:
the folder of the repository  : or do i need to go to a specific folder of the repo

chrivers, 2023-04-10:
Right now your prompt (powershell) is in `Desktop\Github`, but you need to be in the folder you got from github.

You need to be in the main folder of the repository, which is the one that has a file called `requirements.txt`

To change folder, type `cd <name>` to enter `<name>`. Use `cd ..` to go a level up.

Jean0726, 2023-04-10:
I put the repo from github on the desktop, that was wrong ?

chrivers, 2023-04-10:
Nothing wrong with that, as such.

However, where is the `requirements.txt` file located? In your screenshot, your command prompt is in the folder "Github" (which is a folder on your Desktop).

I imagine you have a folder inside "Github", where you have a copy of this repository?

If so, you need to enter that directory in your command prompt.

Let me know if that helps - if not, please try to describe exactly what you downloaded, how you unpacked, and what commands you tried (with screenshots if possible).

Jean0726, 2023-04-10:
ok with the pip3 .....requirements.txt I've done it now ,but now i'm stuck here same problem:

Support for logging in with username and password is not yet supported, but an auth_token can be imported from the saved credentials found in login.json in Ankermake Slicer. See ankerctl.py config import -h for details:

Usage: ankerctl.py config import [OPTIONS] path/to/login.json

  Import printer and account information from login.json

  When run without filename, attempt to auto-detect login.json in default
  install location

Options:
  -h, --help  Show this message and exit.

chrivers, 2023-04-10:
What problem is that?

The good news is that you can start the program now.

Did you read the message that you pasted here? In particular:

```
When run without filename, attempt to auto-detect login.json in default
install location
```

So, try to run `ankerctl.py config import`, and see if that works.

Jean0726, 2023-04-11:
[image]
I have that now and what should I enter now

Jean0726, 2023-04-11:
is that ok now i can start the webserver ?
[image]

Jean0726, 2023-04-11:
ok 
[image]

Jean0726, 2023-04-11:
and must save that now or what do i do now , ps you are more than helpfull

Jean0726, 2023-04-11:
yes i did and followed the instructions also tested the connection to the octoprint is ok but it doesn't print nothing uploads the printer doesn't react. i pressed upload and print but nothing

LazeMSS, 2023-04-11:
What does the shell/prompt say when uploading?

chrivers, 2023-04-16:
Based on your other ticket, it looks like you at least got the installation working, so I'm closing this issue. Feel free to reopen if that's not the case.

closed 2023-04-16 by chrivers