Ankermgmt/ankermake-m5-protocol #47: Cert error when executing mqtt command but not from the ankerctl directory

dmbaio, 2023-04-11:
I get the following error if I run the python file with an mqtt option, but only if I'm not CD'd into that directory.

[image]

chrivers, 2023-04-11:
Thanks for reporting this - I'll see if I can get it fixed soon.

We need to add some code to the python script, to locate the certificate file, regardless of the current working directory

thomasjpatterson, 2023-05-08:
Thanks @dmbaio, we've been furiously working behind the scenes and one of the features that will be upcoming is the ability to use a web interface to select your login.json file for use so you can have it anywhere. Once #59 is complete you'll have this capability.

billyjbryant, 2023-05-21:
This is addressed by #71 and will be released with v1.0.0

thomasjpatterson, 2023-05-21:
Ready to Ship.

closed 2023-05-21 by thomasjpatterson