Ankermgmt/ankermake-m5-protocol #48: Gcode Prompt Output Truncated

dmbaio, 2023-04-12:
As the title says, the output returned from the code prompt, at least for M503 to read settings, seems to grab a section of the middle of the output instead of showing the full output.

chrivers, 2023-04-16:
Hello dmbaio :)

Thanks for reporting this error.

I've looked into it, and it seems like this is a limitation of the printer firmware. It's not clear if there's any way around this, but it doesn't look like it.

We are decoding and displaying pretty much exactly what we receive over MQTT, so unless there's some way to indicate that we want/can handle larger responses, this doesn't seem to be possible.

Which is, of course, a real shame, since it limits what we can do with this.

I'll keep this ticket open for now, until we figure out definitively what's going on.

chrivers, 2023-05-16:
Everything I've looked into, shows that this is, unfortunately, a firmware problem in the printer. We simply cannot get more output than this, until they fix the firmware.

I hope they will eventually fix or improve it, but until they do, there's nothing we can do to fix it. The data just isn't there :-(

I'll close this for now, since it's out of our hands.

closed 2023-05-16 by chrivers