Ankermgmt/ankermake-m5-protocol #49: everything is running now but i can't connect to the printer

Jean0726, 2023-04-13:
webserver test  says ok  but nothing arrives at the printer when I upload it with prusa

chrivers, 2023-04-16:
This can typically happen, if the computer that runs `ankerctl.py` is not on the same wifi as the printer, or if you have some sort of firewall running, that is blocking the traffic.

The official AnkerMake Slicer supports relay mode (where the traffic is sent to AWS (Amazon Cloud) and back to the printer) if the printer cannot be contacted directly.

We don't support relay mode (yet), so this is a limitation for now.

Does that help?

austinrdennis, 2023-04-28:
I get the same issue, but I'm definitely on the same wifi network. Does this not play nice with mesh clients?

chrivers, 2023-04-28:
@austinrdennis can you do a wireshark capture while attempting this? Wireshark should be started before ankerctl, to ensure we see all traffic.

MEvansDCC, 2023-05-03:
I am also having the same issue. I've connected my laptop (running the webserver) and the Anker to the same separate access point. And unfortunately the printer is not doing anything. I can however get a response from the command : 
"python .\ankerctl.py mqtt monitor"
[image]

kivinen, 2023-05-16:
This can also happen when the printers IP-address has changed, for example because the DHCP server decided to assign it different address than before. I think the current code still uses the ip-address it gets from the cloud and it can take days before the cloud backend gets up to date ip-address for the device.

So if you run config show and pppp lan-search and see if the ip-addresses are not same, if so then you need to manually edit the config file to fix the ip-address.

billyjbryant, 2023-05-24:
This typically indicates that you have a communication problem with the printer. Either the printer is on a different network or your local firewall on your computer is preventing you from connecting.

MEvansDCC, 2023-05-24:
Quite right! Failed at the first hurdle 😄 turned off firewall and it worked. So will need to make a rule.

chrivers, 2023-05-24:
@MEvansDCC that's great to hear!

Unfortunately, the pppp protocol is tricky to firewall exactly. But at least, you can allow all UDP traffic, which covers all pppp, instead of allowing all protocols.

closed 2023-05-25 by chrivers