Ankermgmt/ankermake-m5-protocol #51: Changing IP-address of the printer causes issues

kivinen, 2023-04-17:
It seems the ankerctl.py caches the ip-address of the printer to the default.json thus if you use printer in multiple networks you need to manually go to change the ip-address of the .config/ankermake/default.json file to get the ankerctl working on new network. 

Removing the default.json and rerunning ankerctl.py config import login.json does not fix the issue, it seems to be getting the ip-address from some cloud service or something. I have no idea how to get the IP-address updated in the cloud.

'voimakivi (17:45) ~/anker/ankermake-m5-protocol>rm ~/.config/ankerctl/default.json
voimakivi (17:45) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py config show
[*] Loading config from /home/kivinen/.config/ankerctl/default.json

[E] No printers configured. Run 'config import' to populate.
voimakivi (17:45) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py config import ../login.json
[*] Loading cache..
[*] Initializing API..
[*] Requesting profile data..
[*] Requesting printer list..
[*] Requesting pppp keys..
[*] Adding printer [XXXXXXX]
[*] Finished import
voimakivi (17:45) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py config show
[*] Loading config from /home/kivinen/.config/ankerctl/default.json

[*] Account:
    user_id:    XXXX...<REDACTED>
    auth_token: XXXX...<REDACTED>
    email:      kivinen@iki.fi
    region:     EU

[*] Printers:
    duid:      EUPRAKM-XXXX-XXXX
    sn:        XXXX
    ip:        10.21.0.226
    wifi_mac:  XX:XX:XX:XX:XX:XX
    api_hosts: 18.168.14.146, 15.188.180.60, 18.157.241.124
    p2p_hosts: 18.157.241.124, p2p-mk-lon.ankermake.com, p2p-mk-par.ankermake.com
voimakivi (17:45) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py pppp capture-video -m 1m output.h264
[*] Trying connect over pppp
^C
Aborted!
[1]    10764 exit 1     ~/.venvs/anker/bin/python3 ankerctl.py pppp capture-video -m 1m output.h264
voimakivi (17:47) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py pppp lan-search
[*] Printer [EUPRAKM-XXXX-XXXX] is online at 192.168.4.22
voimakivi (17:47) ~/anker/ankermake-m5-protocol>sed 's/10.21.0.226/192.168.4.22/g' -i.orig ~/.config/ankerctl/default.json
voimakivi (17:47) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py config show
[*] Loading config from /home/kivinen/.config/ankerctl/default.json

[*] Account:
    user_id:    XXXX...<REDACTED>
    auth_token: XXXX...<REDACTED>
    email:      kivinen@iki.fi
    region:     EU

[*] Printers:
    duid:      EUPRAKM-XXXX-XXXX
    sn:        XXXX
    ip:        192.168.4.22
    wifi_mac:  XX:XX:XX:XX:XX:XX
    api_hosts: 18.168.14.146, 15.188.180.60, 18.157.241.124
    p2p_hosts: 18.157.241.124, p2p-mk-lon.ankermake.com, p2p-mk-par.ankermake.com
voimakivi (17:48) ~/anker/ankermake-m5-protocol>~/.venvs/anker/bin/python3 ankerctl.py pppp capture-video -m 1m output.h264
[*] Trying connect over pppp
[*] Established pppp connection
1.02Mb [00:02, 375kb/s, size=1.02MB]
[*] Successfully captured 1.02MB video stream into output.h264
voimakivi (17:48) ~/anker/ankermake-m5-protocol>
'

kivinen, 2023-04-17:
It seems that `appapi.query_fdm_list()` returns a corrent wifi ssid, but incorrect ip_addr, so most likely this is issue in the cloud service, not in the ankerctl, but perhaps there should be option that will update the ip_addr based on the ip-address found when using lan-search.

kivinen, 2023-04-18:
Christian Iversen writes:
> Hm, that's an interesting edge case you found. I don't think most people use
> the printer on more than one network, but it's a valid thing to do, of course.
> 
> Just curious, how do you switch between networks? Isn't it a lot of hassle?

I carried the printer to my car, drove to my friends place, carried
the printer inside, reconfigured the wifi using the app. Everything
worked there, and the IP-address I am now getting from the cloud seems
to be coming from there. Then after few days I carried the printer
back to my car, drove home, installed it here again and changed
network back to my home network using app. After that I noticed that
I can't connect to printer anymore using pppp print-file, and started
to investigating. Then noticed that the ip-address is wrong.

I would assume anker will update the ip-address somehow at some point,
but the cloud still seems to be using wrong address. On the other hand
I have not used the ankermake at all, not in my friends place, nor at
home (I did start it, and it did see temperature information, but did
not try to send anything to printer through it yet).

The address returned by the cloud seems to be wrong still, i.e., it
was not fixed after few days... I will monitor the situation and see
if I can see when and what I need to do to get the ip-addresses
changed. The app and ankermake slicer might not even notice that the
ip-address is wrong, as they most likely simply just relay everything
through the cloud...

> I like your idea - we could make lan-search update the IP in the config.
> 
> But maybe a better solution is to build some logic into cli/pppp.py, that can
> help detect the printer More Better(tm).
> 
> Would you like to take a shot at implementing this? I'd be happy to help you
> along the way. It's always nice to have more people involved :)

Unfortunately I can't promise to have any time for this project, I am
way too busy with my current projects. I can see if I have some time
at some point to do something, but can't promise anything... 
-- 
***@***.***

chrivers, 2023-04-19:
I agree, that it's highly surprising how laggy the API is.

And I completely agree, that the anker apps probably work by resorting to relaying. (that *could* also help explain why some people are seeing wildly different performance with the official apps).

There's no rush - if you want to give it a shot at some point, you're more than welcome.

Until then, I'll keep this issue open, so we have some documentation for it.

kivinen, 2023-04-21:
Ok, I now run config import again and now I it got correct ip-address for my printer. So it seemed to take about 4 days to update the address in cloud.

chrivers, 2023-04-21:
That's very good to know - thank you :)

thomasjpatterson, 2023-05-08:
Closing as it looks like this specific issue has been resolved.

closed 2023-05-08 by thomasjpatterson

billyjbryant, 2023-05-08:
> Closing as it looks like this specific issue has been resolved.

I wouldn't say **resolved** but an identified issue that is out of scope (issue exists outside of the control of `ankerctl`)