Ankermgmt/ankermake-m5-protocol #53: Multiple Printer support

just-trey, 2023-05-02:
Currently this solution only supports one printer. If multiple printers are detected it chooses the first one in the array. I would like a solution that supports multiple printers or at least let's us specify one to connect to.

kivinen, 2023-05-05:
I needed this myself as my friend bought ankermake too, and now I have two printers configured. 

There is pull request #55 to implement that.

This is minimal implementation i.e., it refers printers using their indexes, thus saying 

`python3 ankerctl.py -p 1 pppp capture-video -m 1m output.h264`

or 

`python3 ankerctl.py -p 1 webserver run`

to select printer number 1, i.e., your second printer.

billyjbryant, 2023-05-08:
fixed by #55

thomasjpatterson, 2023-05-21:
Ready to Ship

closed 2023-05-21 by thomasjpatterson