Ankermgmt/ankermake-m5-protocol #56: Logs get spammed if printer is powered off

spuder, 2023-05-08:
If you run `ankerctl webserver run` while the printer is powered off, the logs will be spammed with the following message dozens/hundreds of times per second. 

```
OSError: [Errno 64] Host is down
[*] Trying connect over pppp
[E] PPPPService: Failed to start worker: [Errno 64] Host is down. Retrying in 1 second.
Traceback (most recent call last):
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/web/lib/service.py", line 93, in _attempt_start
    self.worker_start()
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/web/service/pppp.py", line 41, in worker_start
    api.connect_lan_search()
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/ppppapi.py", line 264, in connect_lan_search
    self.send(PktLanSearch())
  File "/Users/spuder/Code/github/Ankermgmt/ankermake-m5-protocol/libflagship/ppppapi.py", line 366, in send
    self.sock.sendto(resp, addr or self.addr)
OSError: [Errno 64] Host is down
^C[*] Trying connect over pppp
```
 
Git commit: 3aabda3711cbe77afc02d7fc2801fd294f7bfe34
Git branch: chrivers/webserver-improvments
Environment: python (non-docker, though docker likely has the same issue)

chrivers, 2023-05-09:
Master branch doesn't have proper restart support, and is much less verbose about exceptions.

I agree that it's spammy (and scary looking), but it does work.

I'll add an option to enable backtraces on exceptions, and leave it off by default.

That way, there'll be one line per connection attempt, and nothing more. That should be a good balance between verbosity and feedback.

chrivers, 2023-05-09:
Wait, I might have misunderstood.

You say multiple times *per second*?

That shouldn't be the case - this should occur at most once per second.

I'll do some more digging.

chrivers, 2023-05-09:
Right, I found the error!

Derp... 😅

https://github.com/Ankermgmt/ankermake-m5-protocol/commit/51eaf4c40f8e27bf0c66efd20d9994219e752d2f

I have a branch for this. I'll try to incorporate some stuff useful for @billyjbryant's branch as well (service restart support)

thomasjpatterson, 2023-05-21:
Ready to Ship

closed 2023-05-21 by thomasjpatterson