Ankermgmt/ankermake-m5-protocol #57: Docker unable to resolve dependencies

spuder, 2023-05-08:
When running the docker compose, it produces the following errors

git branch: `chrivers/webserver-improvements`
git commit: 3aabda3711cbe77afc02d7fc2801fd294f7bfe34

`docker compose up --force-recreate --build`

```
Attaching to ankerctl
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
ankerctl  | Traceback (most recent call last):
ankerctl  |   File "/app/ankerctl.py", line 28, in <module>
ankerctl  |     import web
ankerctl  | ModuleNotFoundError: No module named 'web'
ankerctl exited with code 1
^CGracefully stopping... (press Ctrl+C again to force)
Aborting on container exit...
```

spuder, 2023-05-08:
Looks like it is missing web.py dependency https://pypi.org/project/web.py/

## Attempt 1

`docker compose up --force-recreate --build`

requirements.txt

```
web.py==0.51
```

```
ankerctl  |   File "/app/ankerctl.py", line 464, in webserver
ankerctl  |     web.webserver(env.config, host, port, pppp_dump=env.pppp_dump)
ankerctl  |     ^^^^^^^^^^^^^
ankerctl  | AttributeError: module 'web' has no attribute 'webserver'
```

## Attempt 2

`docker compose up --force-recreate --build`

requirements.txt
```python
web.py==0.62
```

```
ankerctl  |   File "/app/ankerctl.py", line 464, in webserver
ankerctl  |     web.webserver(env.config, host, port, pppp_dump=env.pppp_dump)
ankerctl  |     ^^^^^^^^^^^^^
ankerctl  | AttributeError: module 'web' has no attribute 'webserver'
```

---

I'm not sure which dependency it is missing, adding web.py to the requirements.txt isn't fixing the issue.

billyjbryant, 2023-05-08:
Its actually missing the `/web/` folder. 

https://github.com/Ankermgmt/ankermake-m5-protocol/blob/3aabda3711cbe77afc02d7fc2801fd294f7bfe34/Dockerfile#LL22C1-L26C19

 This will be resolved by #52 is merged into https://github.com/Ankermgmt/ankermake-m5-protocol/tree/chrivers/webserver-improvements

thomasjpatterson, 2023-05-18:
Merged into upcoming and no longer seeing issue. Closing ticket

closed 2023-05-18 by thomasjpatterson