Ankermgmt/ankermake-m5-protocol #70: [Bug] PPPPService fails to stop cleanly

billyjbryant, 2023-05-11:
### Description

In testing for the v1.0.0 release, I identified a bug wherein the PPPPService worker crashes, does not restart and will not shutdown when given a shutdown signal, whether that be generated by the UI or the `CTRL+C` sighook. 

See the discord conversation for more details: https://discord.com/channels/1032981123640340530/1088562736633495552/1106000639382802452

#### Logs

```py
^C [*] MqttQueue: Requesting stop
[D] ServiceManager: Shutting down threads..
[D] Service state
[D]   [   1] pppp                 running=True state=RunState.Running wanted=False
[D]   [   2] videoqueue           running=True state=RunState.Running wanted=True
[D]   [   0] mqttqueue            running=True state=RunState.Running wanted=False
[D]   [   0] filetransfer         running=True state=RunState.Stopped wanted=False
[*] PPPPService: Requesting stop
[*] VideoQueue: Requesting stop
[*] MqttQueue: Requesting stop
[D] ServiceManager: Waiting for threads to stop..
[D] Service state
[D] VideoQueue: Stopping worker
[D]   [   1] pppp                 running=True state=RunState.Running wanted=False
[*] PPPPService: Requesting stop
[D]   [   2] videoqueue           running=True state=RunState.Stopping wanted=False
[*] VideoQueue: Worker stopped
[D]   [   0] mqttqueue            running=True state=RunState.Running wanted=False
[D]   [   0] filetransfer         running=True state=RunState.Stopped wanted=False
[D] MqttQueue: Stopping worker
[*] MqttQueue: Worker stopped
^C Exception ignored in atexit callback: <bound method ServiceManager.atexit of <web.lib.service.ServiceManager object at 0x102248070>>
Traceback (most recent call last):
  File "/Users/billybryant/github/personal/ankermake-m5-protocol/web/lib/service.py", line 246, in atexit
    svc.await_stopped()
  File "/Users/billybryant/github/personal/ankermake-m5-protocol/web/lib/service.py", line 220, in await_stopped
    self.idle(timeout=0.1)
  File "/Users/billybryant/github/personal/ankermake-m5-protocol/web/lib/service.py", line 96, in idle
    if self._event.wait(timeout=timeout):
  File "/Users/billybryant/anaconda3/lib/python3.10/threading.py", line 607, in wait
    signaled = self._cond.wait(timeout)
  File "/Users/billybryant/anaconda3/lib/python3.10/threading.py", line 324, in wait
    gotit = waiter.acquire(True, timeout)
KeyboardInterrupt: 
(base) 
```

chrivers, 2023-05-16:
I've collected a number of fixes and improvements for the service handling.

Can you try the experimental branch https://github.com/Ankermgmt/ankermake-m5-protocol/tree/chrivers/service-improvements2 to see if it handles it better?

thomasjpatterson, 2023-05-21:
Ready to Ship

closed 2023-05-21 by thomasjpatterson