Ankermgmt/ankermake-m5-protocol #78: [BUG] Data must be padded to 16 byte boundary in CBC Mode

thomasjpatterson, 2023-05-16:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the bug**
When attempting to run ./ankerctl.py webserver run I encounter a good connection to MQTT and PPPP but I then encounter the following error:

`[E] MqttQueue: Unexpected exception while running worker
Traceback (most recent call last):
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/web/lib/service.py", line 117, in _attempt_run
    self.worker_run(timeout=0.3)
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/web/service/mqtt.py", line 17, in worker_run
    for msg, body in self.client.fetch(timeout=timeout):
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/libflagship/mqttapi.py", line 135, in fetch
    self._mqtt.loop(timeout=timeout)
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 1120, in loop
    return self._loop(timeout)
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/libflagship/mqttapi.py", line 51, in _on_message
    pkt, tail = MqttMsg.parse(msg.payload, key=self._key)
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/libflagship/mqtt.py", line 131, in parse
    body, data = p[:64], mqtt_aes_decrypt(p[64:], key)
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/libflagship/megajank.py", line 30, in mqtt_aes_decrypt
    return aes_cbc_decrypt(cmsg, key, iv)
  File "/Users/username-here/Documents/GitHub/Sublime/ankermake-m5-protocol/libflagship/megajank.py", line 20, in aes_cbc_decrypt
    pmsg = aes.decrypt(cmsg)
  File "/Users/username-here/anaconda3/lib/python3.10/site-packages/Cryptodome/Cipher/_mode_cbc.py", line 246, in decrypt
    raise ValueError("Data must be padded to %d byte boundary in CBC mode" % self.block_size)
ValueError: Data must be padded to 16 byte boundary in CBC mode
[W] MqttQueue: Stopping worker due to exception`

**To Reproduce**
Steps to reproduce the behavior:
1. pip install -r requirements.txt
2. ./ankerctl.py config import for a successful import of login.json
3. ./ankerctl.py webserver run
4. Launch browser to 127.0.0.1:4470
5. See error

**Expected behavior**
No error should occur and I should be able to view at least temperatures on screen (not seeing any of that)

**Desktop (please complete the following information):**
 - OS: Mac M2 Pro
 - Browser: Chrome
 - Version 113.0.5672.92 (Official Build) (arm64)
 - python: 3.10.9

**Additional context**
Add any other context about the problem here.

chrivers, 2023-05-16:
I see you're using the git version - which branch are you on?

thomasjpatterson, 2023-05-16:
The one thing I missed lol. The upcoming/v.1.0.0 branch

chrivers, 2023-05-16:
@thomasjpatterson It seems the mqtt format has changed. I've implemented a check for this in this branch:

https://github.com/Ankermgmt/ankermake-m5-protocol/tree/chrivers/detect-mqtt-format

It'll be merged into the upcoming 1.0.0 branch

thomasjpatterson, 2023-05-16:
Sounds good, I assigned the branch to this issue so once merged, this ticket will auto close.

thomasjpatterson, 2023-05-21:
Ready to Ship

closed 2023-05-21 by thomasjpatterson