Ankermgmt/ankermake-m5-protocol #86: [ENHANCEMENT] Living Documents for Slicers

thomasjpatterson, 2023-05-18:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the solution you'd like**
We currently have a page in the webgui on instructions to connect the Prusa Slicer with AnkerControl. This enhancement request is to add multiple slicer support documentation while converting them to markdown so we can have one document store which would feed the documentation in Github as well as the WebUI

**Acceptance Criteria**
- A user will be able to launch the webui and be provided with documentation on how to connect slicers to AnkerControl: Prusa, Orca, BambuStudio, SuperSlicer, and Cura.
- A user will be able to navigate to the AnkerControl Github and be provided the identical documentation on how to connect slicers to AnkerControl.

chrivers, 2023-05-20:
Another requirement:

 - All material needs to be available in this repository, to make `ankerctl` useable in an offline setting.

chrivers, 2023-05-20:
That was maybe implied, but just to make it clear :)