Ankermgmt/ankermake-m5-protocol #87: [ENHANCEMENT] MainSail Support

thomasjpatterson, 2023-05-18:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the solution you'd like**
To provide an even richer experience for advanced users, we shall provide MainSail support for AnkerControl such that we will become a "middleware" similar to MoonRaker to control the AnkerMake M5.

**Acceptance Criteria**
- User should be able to launch AnkerControl and be provided steps on how to add it to MainSail.
MainSail Requirements
- Users should be able to view temperatures of the M5
- Users should be able to see bed leveling status of the M5
- Users should be able to send gcode through MainSail to their printer
- Users should be able to stop their printer through MainSail
- Users should be able to initiate an emergency stop to their printer through MainSail
- Users should be able to view their camera through MainSail.

chrivers, 2023-05-20:
In the experimental mainsail branch, we have the following features:

 - [x] Users should be able to view temperatures of the M5
 - [ ] Users should be able to see bed leveling status of the M5
 - [x] Users should be able to send gcode through MainSail to their printer
 - [ ] Users should be able to stop their printer through MainSail
 - [ ] Users should be able to initiate an emergency stop to their printer through MainSail
 - [x] Users should be able to view their camera through MainSail.

Now, an "Emergency stop" feature is not something I'm sure has an M5 equivalent. But we could make the E-stop feature send some kind of stop command, to at least have a best-effort option.

Bed leveling is unlikely to work. Not because we can't provide the Mainsail UI with the data, but because the M5 can't provide the data to us :(

We can request the data through gcode (over mqtt), but because the packets are truncated at 512 bytes, we don't actually get the data. So it seems we are out of luck, unless Anker fixes the firmware to actually support gcode commands.

ravenspeak, 2023-12-21:
@chrivers  do you have any quick guide on how to make the experimental mainsail branch work? Do I just run the webserver and add the printer in mainsail using the webserver's IP and port?