Ankermgmt/ankermake-m5-protocol #91: [BUG] The Copy to Clipboard buttons do not work

billyjbryant, 2023-05-20:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the bug**
When clicking on the copy to clipboard buttons for the OctoPrint Hostname and the Login.json path, you are presented with the following error in Console: 

```
Uncaught TypeError: Cannot read properties of undefined (reading 'writeText')
    at HTMLButtonElement.<anonymous> (ankersrv.js:23:33)
    at HTMLButtonElement.et (cash.min.js:1:16170)
(anonymous) @ ankersrv.js:23
et @ cash.min.js:1
```

The clipboard does not get written to. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Instructions' or 'Setup'
2. Click on the clipboard button
4. Open the Developer's Console
5. See error

**Expected behavior**
Clicking on the clipboard button should copy the preceding information presented into the clipboard

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows
 - Browser [e.g. chrome, safari] Edge
 - Version [e.g. 22] Version 113.0.1774.42 (Official build) (64-bit)

**Additional context**
This issue is in `upcoming/v1.0.0`

chrivers, 2023-05-20:
I tested in Firefox and Chrome, and they both work on that branch.

The backtrace you provided has this on `ankersrv.js:23`:

```js
            navigator.clipboard.writeText(value);
```

So it seems like edge just doesn't allow clipboard access. From what I could find, Edge 81 and upwards have restricted access to the clipboard. Can you test if this guide makes it work?

https://learn.microsoft.com/en-us/answers/questions/766718/how-do-i-unblock-the-clipboard-for-a-website

(that's not necessarily a solution, but it would give us a clear indication if we understand the problem)

billyjbryant, 2023-05-20:
Interesting. It worked during locally when I was working on it before. I wonder if Edge treats it differently if it is coming from localhost vs a non-local IP address. Let me pull down the repo locally and run it to see if it is the same.

chrivers, 2023-05-20:
That could very well be the case.

From some googling, it appears that we can request access to the clipboard from the browser, but I'm not quite sure how that works. Let me know what you find.

billyjbryant, 2023-05-20:
Ok, I've found the issue, In Chromium browsers, you cannot access the clipboard over insecure sites unless said insecure site is localhost. 

So in my case, I run `ankerctl` via Docker on my NAS and it has a non-localhost IP address of 192.168.5.192. Since the site is insecure, the clipboard is not available. But when I run a local Docker or through python directly, the site is hosted at 127.0.0.1 and the clipboard is available. 

There is a way to bypass this setting for the website. 

### Enable Clipboard for Local Sites in Chromium
1. Open the browser
2. Navigate to [edge://flags/#unsafely-treat-insecure-origin-as-secure](edge://flags/#unsafely-treat-insecure-origin-as-secure)
3. Locate the setting titled **`Insecure origins treated as secure`**
[image]
4. Add your container's IP address as `http://<IPAddress>:<Port>` (IE. If your Ankerctl IP is `192.168.1.5` and running on the default port (`4470`), then you would enter `http://192.168.1.5:4470`)
5. Change the dropdown from `Disabled` to `Enabled`
6. A confirmation bar will appear at the bottom of the screen, click the `Restart` button to restart the browser.

chrivers, 2023-05-20:
Wow, that's a lot more involved than I would have guessed :sweat_smile:

The explanation makes sense - sadly, the fix is greater than the inconvenience of having to press ctrl-c, or right-clicking and pressing "Copy"... :-)

I have one idea though - right now we are setting up the clipboard icons with javascript.

We can easily detect if the support is missing, and in that case hide the buttons instead. That's a fairly simple solution, and should minimize the potential for confusion. How about it?

billyjbryant, 2023-05-20:
Sounds like a good plan to me. Especially since there are only two times that you may need to copy/paste data from the site. Doesn't really make sense to provide the "workaround" in documentation for a simple thing.

chrivers, 2023-05-20:
I've implemented this fix in `chrivers/infrastructure-improvements`. Can you try out that branch, and see if it works as expected?

billyjbryant, 2023-05-20:
That works. 
[image]

chrivers, 2023-05-20:
Excellent - I'll get that branch merged then :+1:

billyjbryant, 2023-05-21:
Closing this as resolved.

closed 2023-05-21 by billyjbryant