Ankermgmt/ankermake-m5-protocol #92: [ENHANCEMENT] PyPi release

kvanzuijlen, 2023-05-20:
**Describe the solution you'd like**
I'd like to take a look if I can start working on an implementation for Home Assistant. It would be nice to have a PyPi release for the underlying library.

**Describe alternatives you've considered**
Home Assistant requires the implementation to be separated in a separate dependency. Since this project is also in active development a PyPi package would allow for easier upgrades.

**Additional context**
I have written 2 other core Home Assistant integrations before, currently working on a third. If no one has implemented this in the meantime I'd like this to be the 4th.

chrivers, 2023-05-20:
Hello @kvanzuijlen, and welcome to the project :)

We would also very much like to see Home Assistant integration - it would be an awesome thing to have.

This printer is particularly complicated to interface with, for a few reasons. Let me go through them step by step:

 - Multiple APIs
 - Proprietary APIs
 - Connection limits

# Multiple APIs

The printer uses 4(!) different APIs for communication with the outside world:

 - HTTPS (Fairly ordinary HTTPS API)
 - MQTT (Non-standard use of MQTT)
 - PPPP (Proprietary peer-to-peer protocol)
 - BLE GATT (Bluetooth Generic Attribute protocol - used for providing Wifi credentials on first setup)

For full access to the printer (as full as the printer allows, anyway), you need to support the first three protocols.

The library present in this repository, `libflagship`, provides the low-level implementations for the first three protocols.

However, since the APIs are interdependent, a non-trivial amount of code is required to handle the various authentication tokens, crypto keys, device keys and other credentials. This is what the CLI tool (and the upcoming embedded webserver) implement, to provide a smooth user experience.

# Proprietary APIs

Sadle, all the APIs are proprietary and heavily obfuscated, to make it as hard as possible to interact with the printer. A great deal of work has gone into reversing these APIs, and provide a somewhat sane interface to developers. For a sample of some of the things going on, take a look here:

https://github.com/Ankermgmt/ankermake-m5-protocol/blob/master/libflagship/megajank.py

This, at least, is handled almost entirely by `libflagship`.

# Connection limits

So this one is the real kicker. There are 2 fundamental limits that are in the way:

 - Single auth token
 - Single printer connection

## Single auth token

The HTTPS API provides an `auth_token`, which enables the holder to request other authentication information needed to connect MQTT and PPPP. The API only allows a single `auth_token` to be valid at any one time.

Even a simple case, like having 2 installs of the official AnkerMake Slicer, will run into this problem. Any time you use one, the other will require you to login again.

For this reason, `ankerctl` (from this repository), can import the AnkerMake Slicer credentials, meaning they can share the `auth_token`, as to not run into these same issues.

## Single printer connection

On top of this, the printer *itself* only allows a single concurrent PPPP connection. The PPPP connection is what provides file uploads, as well as control over the video feed.

A Home Assistant integration may or may not want/require a PPPP connection, which would make the problem a lot simpler.

# Way forward

So, to summarize. I see 2 potential ways forward:

## Option 1: New software, using `libflagship` independently

A Home Assistant integration could use code from this repository in some manner (git submodule, python package, etc), and support importing the AnkerMake Slicer `login.json` file. This can then be used to connect to MQTT, which will allow for print job status reporting, as well as basic information such as temperature of the nozzle and print bed.

As long as only MQTT is used, this will not block other programs from communicating with the printer.

## Option 2: Running `ankerctl` in web server mode, and using its APIs

The (soon to be released) version 1.0 of `ankerctl` (from this repo) has a built-in webserver, which provides websocket APIs for print job status, video feed, file upload, etc.

The webserver itself has a user-friendly interface for uploading a `login.json` from AnkerMake Slicer, and provides a dashboard of its own.

However, `ankerctl` server can support any (reasonable) number of client connections. This means that a Home Assistant plugin that talks to `ankerctl` could be simpler, while also supporting more advanced features.

Most importantly, it would allow complete integration with Home Assistant, while also allowing other clients to work at the same time.

The webserver comes bundled in a docker container, so it should be relatively simple to set up. The Home Assistant integration just needs to be pointed to the docker, and then things should work from there.

We'd be happy to work with you on this - let me know what you think.

thomasjpatterson, 2023-05-20:
@kvanzuijlen we have a discord if you would like to join it https://discord.gg/HB2eRhDe to discuss more about how to progress on this initiative.

kvanzuijlen, 2023-05-20:
Hello @chrivers,

I think both solutions could be implemented. I'd focus on the second one first though, since it sounds like this is the easiest one to implement. The integration folders in home assistant core could be named ankermake and ankermake_local (local being the one that requires the webserver). Both will then show up under the same AnkerMake entry in the list of integrations.

I'll join the Discord shortly, but will be focussing on the other integration first.

Thanks for all the great work! It's sad to see Anker being so secretive (I also have a Robovac X8 I'd like to better integrate in Home Assistant).

chrivers, 2023-05-20:
Both can absolutely be done, and I agree that the second is probably the easiest to implement.

Perhaps the integrations could be called `ankermake` and `ankerctl`, since the second one really integrates with `ankerctl`, and has no other dependencies. It's up to you, of course.

Looking forward to seeing you in the discord when you have time.

rcourtman, 2024-02-20:
Big thanks for all the effort you've put into ankerctl. It's shaping up to be a very useful tool for these line of printers. 

I've been thinking about how we might simplify integrating the AnkerMake printers with Home Assistant, especially given the challenges you've outlined with the APIs and connection limits. Maybe we can use a virtual serial port to link OctoPrint to ankerctl.

### Here's what I'm thinking:

Use socat to create a virtual serial port on the machine that's running OctoPrint and ankerctl.
Write a simple bridge script that passes commands from OctoPrint through ankerctl to the printer, and vice versa.

### Why this could work:

It keeps things simple by using ankerctl for the complex API stuff.
OctoPrint already has a Home Assistant integration, so if we can get OctoPrint to talk to the printer via ankerctl, we're golden. No need to build a lot from scratch for Home Assistant.
It's a flexible approach that could potentially be used with any printers that communicates via G-code.

### Things to consider:

The key is making sure the bridge script is reliable and fast.
Security is important, as always.

Would love to hear your thoughts on this. It seems like a straightforward workaround that leverages existing tools and integrations.

Thanks!

[image]