Ankermgmt/ankermake-m5-protocol #97: [BUG] Multi-printer support broke running without config

chrivers, 2023-05-21:
Fix coming

billyjbryant, 2023-05-21:
Do you have details of the bug? I run without config to test the import of the webserver ui and I don't see issues.

chrivers, 2023-05-21:
Since we merged multi-printer support, this happens for `webserver run` without config:

```session
[W] No printers found in config. Please upload configuration using the webserver or 'ankerctl.py config import'
Traceback (most recent call last):
  File "/foo/ankerctl.py", line 488, in <module>
    main()
  File "/usr/lib/python3/dist-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/foo/ankerctl.py", line 484, in webserver
    web.webserver(env.config, env.printer_index, host, port, env.insecure, pppp_dump=env.pppp_dump)
  File "/foo/web/__init__.py", line 263, in webserver
    if printer_index >= len(cfg.printers):
                            ^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'printers'
[*] ServiceManager: Shutdown complete
```

We check if the printer index is valid, without checking if we have any printer config at all.

I have a fix incoming, just need a moment to push :)

chrivers, 2023-05-23:
Fixed in upcoming.

commit f081ddf6 (2023-05-21): * Before checking printer index, ensure we have printer config at all (fixes #97)
closed 2023-05-23 by chrivers