Ankermgmt/ankermake-m5-protocol #101: [BUG] PPPP Service fails to connect on webserver

billyjbryant, 2023-05-22:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the bug**
It appears that the fix to the `pppp lan-search` bug, has caused the `webserver` to have an issue when attempting to connect to the PPPP and Video microservices. 

This causes the video to not connect and the `PPPP` badge connects and disconnects repeatedly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./ankerctl.py webserver run`
2. Browse to the webserver (http://127.0.0.1:4470)
3. Go to the console output
4. See error:

```bash
[*] Trying connect to printer Vulcan (USPRAKM-008574-TPUPS) over pppp using ip 192.168.4.145
[D] TX  --> PktLanSearch()
[D] RX <--  PktPunchPkt(duid=Duid(prefix='USPRAKM', serial=8574, check='TPUPS'))
[D] TX  --> PktClose()
[D] TX  --> PktP2pRdy(duid=Duid(prefix='USPRAKM', serial=8574, check='TPUPS'))
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] RX <--  PktClose()
[E] CLOSE
[E] PPPPService: Failed to start worker: . Retrying in 1 second.
Traceback (most recent call last):
  File "/mnt/c/Users/billy/Projects/ankermake-m5-protocol/web/lib/service.py", line 103, in _attempt_start
    self.worker_start()
  File "/mnt/c/Users/billy/Projects/ankermake-m5-protocol/web/service/pppp.py", line 46, in worker_start
    api.process(msg)
  File "/mnt/c/Users/billy/Projects/ankermake-m5-protocol/libflagship/ppppapi.py", line 301, in process
    raise ConnectionResetError
ConnectionResetError
```

**Expected behavior**
The expected behavior is for the PPPP service to connect and video stream to start. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows & WSL Ubuntu
 - Browser [e.g. chrome, safari] Edge
 - Version [e.g. 22] 

**Additional context**
This issue exists in upcoming/v1.0.0

billyjbryant, 2023-05-22:
Actually, this appears to only happen if the video stream is being accessed elsewhere. Not related to the PPPP fix. Maybe we should figure out if it is possible to present a "prettier" error message in this case.

chrivers, 2023-05-22:
I agree; In general, trying to present more accurate and usable status feedback to the user will be a priority for v1.1 and beyond.

What we have now is slightly better than nothing, but it does tend to show green status in many cases where communication is not really working.

chrivers, 2023-05-22:
The printer itself only supports a single video stream, but `ankerctl` supports any (reasonable) number of viewers.

As you point out yourself, this isn't related to the recent pppp fix, so I'm going to close this ticket. We'll get back to improving the feedback after v1 launch :)

closed 2023-05-22 by chrivers