Ankermgmt/ankermake-m5-protocol #109: There should be an option to update the printer ip-number in default.json based on lan-search results

kivinen, 2023-05-23:
This is related to the issue #51 i.e., if the printer IP-number is changed it can take days until the cloud backend will get updated, and even then the default.json is only updated if you rerun the config import.

**Describe the solution you'd like**

We should have 

```
ankerctl.py pppp lan-search --update-ip 
```

command or similar, that would do lan-search for the printer, and if it finds the printer, it should go and check the configuration with matching printer, and if found it should update the ip-number in the config to match the ip-number returned by lan-search.

**Describe alternatives you've considered**

Other alternative what I am did do myself was to use lan-search to find the ip-number and then manually edit the default.json file with editor and updated the ip-number there.

kivinen, 2023-05-23:
At some point the webserver and actually any pppp connection which requires direct connection to printer, should automatically search for printer if the printer times out, and if they find matching printer from some other ip, start using that ip-address instead of the one stored in the config.

For example now it could happen that you are running webserver on rasberry pi and use that to send prints to printer, but if the printer is restarted and gets new ip-address assigned by the dhcp-server, you suddenly need to go and modify the config to get webserver to use new ip-address.

billyjbryant, 2023-05-24:
> but if the printer is restarted and gets new ip-address assigned by the dhcp-server, you suddenly need to go and modify the config to get webserver to use new ip-address.

Simple restarts of the printer are typically not enough to trigger a new IP address unless the printer is off long enough for its lease to expire and another device to come in and snag the lease. 

That said, this is not a bad feature to have.

chrivers, 2023-05-24:
When searching for a printer, we send out a PKT_LAN_SEARCH on broadcast.

I think we can just make the api use whatever IP the printer responds from, as the ip to connect to. If that works, we wouldn't have to trust the saved ip at all. I'll do some experiments..

kivinen, 2023-05-24:
I usually keep my printer off all the time unless I am printing, but I do have static IP-address configure for it in my DHCP server. I would assume others will do same, and if their DHCP lease times are in order of hours the printer might get new addresses every now and then depending how much other devices they have connecting their network. 

And I think we should most likely use the ip-address directly first, as then if someone has printer in different network they can perhaps open suitable firewall rules to allow any udp to go printers ip-address (and allow any response udp packets to come back). I for example do have firewall between my ethernet network and wifi network, so I need to do some firewall plugging to get pppp to work from ethernet to wifi.

We should fall back to do broadcast based search if the printer does not respond with the IP-address we have in configuration.

treitmayr, 2024-02-18:
Implemented in PR #150.