Ankermgmt/ankermake-m5-protocol #112: [ENHANCEMENT] Print from Web UI

billyjbryant, 2023-05-24:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the solution you'd like**
Provide a way to upload gcode files from the WebUI

**Describe alternatives you've considered**
Uploading through a Slicer or the CLI

**Additional context**
Add any other context or screenshots about the feature request here.

chrivers, 2023-05-24:
That should be pretty easy, since we already have a http api for print job upload.

I think the main challenge is actually figuring out how to integrate it into the ui.

Maybe we talk about all the upcoming ideas for features on the discord, and see if we can't come up with a plan for an overall ui design to target :)

billyjbryant, 2023-05-24:
I implemented this in #113 feel free to comment on how I included it :)

chrivers, 2023-05-28:
Fixed in v1.1 :)