Ankermgmt/ankermake-m5-protocol #127: [BUG] PPPP connection drops out after a short delay on webserver startup.

austinrdennis, 2023-07-03:
**Describe the bug**
A similar problem to the one described in #49. I'm having connection problems over PPPP, but only after it works flawlessly for anywhere from 5-45 seconds(ish) before the video stream freezes and the printer stops responding to PPPP commands.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the webserver via python.
2. Open the webpage and allow it to load.
3. Wait for the video stream to freeze.

**Expected behavior**
The PPPP connection continues to function once established.

**Screenshots**
[Screenshots gallery.](https://imgur.com/a/btkbYFb)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Chrome (64-bit)
 - **Versions:** Chrome: 114.0.5735.199 (Official Build), AnkerCtl: 1.0.0, Windows: 22621.1848, Python: 3.11.4

**Additional context**

- There are no signs of trouble in the console, only the periodic MQTT topic messages. 
- The webserver needs to be killed (hard killed) and restarted in order to get PPPP running again, but the problem happens every time the webserver is started and the browser page is loaded. Refreshing services via the webpage does not fix it.
- Thing's I've tried already: forwarding the ports given by @chrivers on my router and disabling Windows firewall. These steps allowed the PPPP connection to be established in the first place only after both steps were complete.

jk-andersen, 2023-07-14:
I am having the same issue running on docker. need to hard kill the process or restart the container

HakonHarnes, 2023-07-24:
I've also been encountering the same issue where the PPPP connection drops out after a short delay. Restarting the ankerctl service resolves the issue temporarily for me. However, after a few service restarts, the PPPP connection stops working completely, and the Ankermake app displays an _"Unable to play real-time video (10000)"_ error. This error is resolved when the printer is turned off and back on.

This behavior suggests that the PPPP connection might not be properly released by ankerctl, effectively resulting in a build-up of active connections. It seems plausible that the printer might have a limit on the number of active PPPP connections it can handle at a time. If these connections are not released correctly, we may exhaust this limit, which would explain why the connection drops and why the issue resolves upon restarting the printer (which would forcibly close these connections).

@chrivers could you shed some light on how the PPPP connection is terminated? I'm not familiar with this but I might look into resolving the issue.

z-master42, 2023-12-14:
I also have a problem with PPPP. However, when I upload my login.json, the wrong IP address and MAC address are extracted. What could be the reason for this?
MQTT and CTRL are green.

austinrdennis, 2025-03-09:
Project is dead.

closed 2025-03-09 by austinrdennis