Ankermgmt/ankermake-m5-protocol #128: An issue with AnkerCTL Window (post-install and post config) popping up a blank screen then vanishing[BUG]

Rascal2023, 2023-07-12:
So I have the m5 Printer obviously. LOL

I've gone through the entire process of installation and importing the configuration. Everything has worked (with some struggles, but got them figured out)

NOW, When I type ./Ankerctl.py webserver run 

It pops up a new "blank" screen for just a second and then it vanishes. Nothing else happens.
I try to go to http://localhost:4470/ and it just can't load the page.

I've tried numerous times. What am I doing wrong?

Rascal2023, 2023-07-12:
I've also tried to do the Config show command and the same thing happens.

Rascal2023, 2023-07-12:
I tried to reimport the config file and now config show is working. But when I do the webserver run command, it still pops up a blank black screen and then it goes away. Still unable to open the site.

billyjbryant, 2023-07-13:
What version of python are you using? `python -v`

Can you also run `python -m pip check` to ensure that you have no broken dependencies?

Rascal2023, 2023-07-13:
When I go to the folder where I have the Ankerctl.py application in Terminal and type Python -m pip, this is what I get 
[image]

Rascal2023, 2023-07-13:
I'm also using the newest Python 3.11

Rascal2023, 2023-07-13:
No broken requirements found is the response I got with Python 3.11. I'm gonna try downloading Python 3.9

billyjbryant, 2023-07-17:
The minimum required version is 3.10. Ankerctl will not run on Python 3.9.

billyjbryant, 2023-07-17:
@Rascal2023 what happens when you run `python .\ankerctl.py -v webserver run`?