Ankermgmt/ankermake-m5-protocol #130: Home assistant integration

cunninr2, 2023-09-14:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the solution you'd like**
Excellent work.  I'm really interested in getting status monitoring of the Ankermake M5  into home assistant so I can monitor from afar and auto shutdown.  Probably only needs a few status parameters such as printing, job completed, bed/nozzle temps. start/stop printing.  The icing on the cake would be a camera feed for sure. keep up the good work.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

BowlesCR, 2023-09-17:
Despite the title, #92 seems to be working in this same direction -- you may want to see about coordinating efforts.

Phoenix-DH, 2024-04-24:
I would also be ninteressted in a HA integration.
Thanks ✌🏻

JayRHa, 2024-06-23:
Please work on this#

sondregronas, 2024-07-07:
Hi - I've been working on a Home Assistant integration that seems to have now reached a usable stage, which is available here: https://github.com/sondregronas/ankermake-hass-component and can be installed via hacs.

Note: I've only tested with ankerctl in the v1.1 branch, and I'm not sure how stable it is yet as finding errors or debugging takes a lot of time

I'm also open to moving said integration over to the Ankermgmt organization if it's of any interest. (https://github.com/Ankermgmt/ankermgmt-hassio-addons/issues/5)

Phoenix-DH, 2024-07-07:
Hey,
this sounds great! Thanks.
Does that mean that in addition to your hacs also an ankerctl instance must be running? Anything else needrd on HA? MQTT Broker?

But where to find ankerctl for HA? Am I blind?

Thanks

sondregronas, 2024-07-07:
You just need an instance of ankerctl running, yes. Once you've installed via HACS / imported to `custom_components` you should find it under "Add Integration" in the `Devices & Services` dashboard. There's no need for an MQTT broker or any other dependencies as it uses ankerctl's own provided webserver (`/ws/mqtt` endpoint)

[image]

Phoenix-DH, 2024-07-07:
Thanks!
Just added your repo to HACS and the Addon to HA.
Will try next time when I use the printer.

Thanks.

sondregronas, 2024-07-10:
You need ankerctl running, I use this fork in docker for the time being: https://github.com/sondregronas/ankermake-m5-protocol/tree/patch-exiles-1.1-auto-restart-on-failure, but I presume the home assistant version should work.

The integration only needs the URL of an ankerctl instance, it doesn't matter if its running via home assistant or not.

Phoenix-DH, 2024-07-10:
Sure, but i do not want to run something in parallel.
I think I found in one orher thread maybe a solution:
https://github.com/Ankermgmt/ankermake-m5-protocol/issues/161

sondregronas, 2024-07-10:
@Phoenix-DH 

https://github.com/sondregronas/ankermgmt-hassio-addons - try replacing the main addon with this (remove the old repo from AddOn repos & replace with [the new one](https://my.home-assistant.io/redirect/supervisor_add_addon_repository/?repository_url=https%3A%2F%2Fgithub.com%2Fsondregronas%2Fankermgmt-hassio-addons) - works for me :smile: (`<hass-ip>:4470`, ingress does not work on mine either)

This is the same Dockerfile as in my fork - which has the same changes as the fork/pr you linked, **_though it is not a long term solution. Once merged I'll most likely remove it.. (or just update it to the main repo's latest release)_**

I'd recommend creating an automation to restart the AddOn every 2 hours, if you want a reliable videostream.

Phoenix-DH, 2024-07-10:
Thanks!
Will try next days. I have only a M5C without video.

Phoenix-DH, 2024-07-10:
Hey, will try it tomorrow! Thanks!

sondregronas, 2024-07-10:
Should be working now **_(requires a rebuild of the addon)_** - I'm not 100% certain the M5C is supported, from what I can tell it should?

Phoenix-DH, 2024-07-11:
Thanks!
I installed the latest and there is a login windows.
Will do the rest of testing today.

Phoenix-DH, 2024-07-11:
Login worked, will install the plugin.
Also the upgrade to v1.0.2. of the addon was working.