Ankermgmt/ankermake-m5-protocol #131: [BUG]

Maddin4000, 2023-10-18:
Don't Show Temperature or Print Progress

MacBook Air 2020 M1
Safari Browser

billyjbryant, 2023-10-18:
Can you expand on what exactly the bug here is?

Maddin4000, 2023-10-18:
It don’t show any time for Print or temperature  

only I see the Camera Live picture

billyjbryant, 2023-10-18:
Can you share more information about your setup? 

* Are you using the docker container?
* Are you on windows|mac|linux?
* What version of Python are you using?
* Is your printer on the local/same network as your computer?
* What steps can be taken to reproduce the problem?
* Do you have any screenshots of the issue you are having?

Maddin4000, 2023-10-19:
Hi, no Docker
Mac Sonoma
Python 3.12
Printer a the same Network
stehst i don't no
[image]

billyjbryant, 2023-10-19:
MQTT status doesn't appear to be connected, do you see any errors in the logs when you are printing? I would definitely say a lack of MQTT communication is at fault for that data being missing. 

You can also go to the Setup tab and click the restart button to restart the services and reload ankerctl.

Maddin4000, 2023-10-20:
reloud dont fix it

screenshot without printing

[image]

billyjbryant, 2023-10-21:
What about the console logs in the browser? Can I see them as well?

Maddin4000, 2023-10-25:
> What about the console logs in the browser? Can I see them as well?

which do you mean?

billyjbryant, 2023-10-27:
Accessing the browser console (typically through `F12` to get into developer tools). 

Also, Can you advise what firmware version you are on with the AnkerMake and which model of printer you have.

Maddin4000, 2023-10-27:
Ok, Ankermake m5, SW 

V3.0.57_3.0.74

Lieben Gruß Martin HilbrandtAm 27.10.23 um 18:59 schrieb Billy Bryant
                
            
            
                
                    Von: "Billy Bryant" ***@***.***>Datum: 27. Oktober 2023An: "Ankermgmt/ankermake-m5-protocol" ***@***.***>Cc: "Maddin4000" ***@***.***>,"Author" ***@***.***>Betreff: Re: [Ankermgmt/ankermake-m5-protocol] [BUG] (Issue #131)
                
                
Accessing the browser console (typically through F12 to get into developer tools).
Also, Can you advise what firmware version you are on with the AnkerMake and which model of printer you have.

Maddin4000, 2023-10-27:
`<!DOCTYPE html>
<html data-bs-theme="dark" lang="en">
    <head>
        <meta charset="utf-8" />
        <meta content="width=device-width, initial-scale=1" name="viewport" />
        <title>ankerctl - AnkerMake M5 (AK7ZRM0C44200218)</title>
        <link rel="icon" href="/static/img/logo.svg">
        <link rel="stylesheet" href="/static/vendor/bootstrap-5.3.0-alpha3-dist/css/bootstrap.min.css" />
        <link rel="stylesheet" href="/static/vendor/bootstrap-icons-1.10.5/font/bootstrap-icons.min.css" />
        <link rel="stylesheet" href="/static/ankersrv.css" />
    </head>
    <body>
        
        <div class="bg-dark bg-gradient mb-3">
            <header class="container d-flex align-items-center py-2">
                [image]
                    <span class="text-body-emphasis align-self-baseline fs-4">ankerctl</span>
                </a>
                <ul class="nav nav-pills align-items-baseline" id="tab" role="tablist">
                    
    
        
    <li class="nav-item" role="presentation">
        <button
            class="btn btn-outline-primary nav-link active"
            id="home-tab"
            data-bs-toggle="pill"
            data-bs-target="#home"
            type="button"
            role="tab"
            aria-controls="home"
            aria-selected="true">
            
            <i class="bi-house px-1 "></i>Home
        
        </button>
    </li>

        
    <li class="nav-item" role="presentation">
        <button
            class="btn btn-outline-primary nav-link"
            id="instructions-tab"
            data-bs-toggle="pill"
            data-bs-target="#instructions"
            type="button"
            role="tab"
            aria-controls="home"
            aria-selected="false">
            
            <i class="bi-list-check px-1 "></i>Instructions
        
        </button>
    </li>

    

    
    <li class="nav-item" role="presentation">
        <button
            class="btn btn-outline-primary nav-link"
            id="setup-tab"
            data-bs-toggle="pill"
            data-bs-target="#setup"
            type="button"
            role="tab"
            aria-controls="home"
            aria-selected="false">
            
        <i class="bi-gear-fill px-1 "></i>Setup
[... 426 lines cut ...]
    class="text-decoration-none"
    href="https://github.com/Ankermgmt/ankermake-m5-protocol"
    rel="noopener noreferrer nofollow"
    target="_blank"
    >
    ankerctl
</a>
<br />
<a
    href="https://github.com/Ankermgmt/ankermake-m5-protocol"
    class="btn btn-link btn-sm"
    rel="noopener noreferrer nofollow"
    target="_blank"
    >
    <i class="bi-github px-1 "></i>View on Github
</a>
                
            </div>
        </footer>
        <script src="/static/vendor/bootstrap-5.3.0-alpha3-dist/js/bootstrap.bundle.min.js"></script>
        <script src="/static/vendor/cash.min.js"></script>
        <script src="/static/vendor/jmuxer.min.js"></script>
        <script src="/static/ankersrv.js"></script>
    </body>
</html>`

Maddin4000, 2023-10-27:
you mean this?

anklaa, 2023-10-29:
Hi dudes,

same issue over here...

I have the same issue when I am using docker and the native python script aswell.

The python script throws the following errors:

`ndle.min.js.map HTTP/1.1" 304 -
[*] VideoQueue: Requesting start
[*] PPPPService: Requesting start
[*] Trying connect to printer AnkerMake M5 (EUPRAKM-002786-KMKMS) over pppp using ip 192.168.0.185
[*] 127.0.0.1 - - [29/Oct/2023 20:43:22] "GET /static/img/logo.svg HTTP/1.1" 304 -
[*] Established pppp connection
[*] PPPPService: Worker started
[*] VideoQueue: Worker started
[E] Failed to decode mqtt message: expected set() but found {20, 205}
[E] Failed to decode mqtt message: expected set() but found {43, 11}
[E] Failed to decode mqtt message: expected set() but found {225, 29}
[E] Failed to decode mqtt message: expected set() but found {129, 12}
[E] Failed to decode mqtt message: expected set() but found {153, 17}
[E] Failed to decode mqtt message: expected set() but found {18, 92}
[E] Failed to decode mqtt message: expected set() but found {2, 183}
[E] Failed to decode mqtt message: expected set() but found {1, 214}
[E] Failed to decode mqtt message: expected set() but found {126}
[E] Failed to decode mqtt message: expected set() but found {227, 37}
[E] Failed to decode mqtt message: expected set() but found {39}
[E] Failed to decode mqtt message: expected set() but found {16, 131}
[E] Failed to decode mqtt message: expected set() but found {9, 35}
[E] Failed to decode mqtt message: expected set() but found {100, 28}
[E] Failed to decode mqtt message: expected set() but found {2, 30}
[E] Failed to decode mqtt message: expected set() but found {42, 35}
[E] Failed to decode mqtt message: expected set() but found {244, 4}
[E] Failed to decode mqtt message: expected set() but found {11, 95}
[E] Failed to decode mqtt message: expected set() but found {21, 103}
[E] Failed to decode mqtt message: expected set() but found {2, 244}
[E] Failed to decode mqtt message: expected set() but found {17, 27}
[E] Failed to decode mqtt message: expected set() but found {32, 82}
[E] Failed to decode mqtt message: expected set() but found {233, 27}
[E] Failed to decode mqtt message: expected set() but found {16, 122}
[E] Failed to decode mqtt message: expected set() but found {139, 12}
[E] Failed to decode mqtt message: expected set() but found {32, 200}
[E] Failed to decode mqtt message: expected set() but found {24, 83}
[E] Failed to decode mqtt message: expected set() but found {161, 20}
[E] Failed to decode mqtt message: expected set() but found {5, 55}
[E] Failed to decode mqtt message: expected set() but found {234}`

The console of the browser throws only:

[image]

martinmcu, 2023-11-01:
I am seeing similar behavior. Using the native python script on Linux Mint 21 and Firefox as my browser:

[image]
[image]

I have an M5 running version 3.0.57_3.0.74.

I have run with the following code modification:
```
diff --git a/libflagship/mqttapi.py b/libflagship/mqttapi.py
index 13dc662..0b6d2a1 100644
--- a/libflagship/mqttapi.py
+++ b/libflagship/mqttapi.py
@@ -52,6 +52,7 @@ def _on_message(self, client, userdata, msg):
             pkt, tail = MqttMsg.parse(msg.payload, key=self._key)
         except Exception as E:
             log.error(f"Failed to decode mqtt message: {E}")
+            log.info(f"msg.payload: {msg.payload}")
             return
 
         data = json.loads(pkt.data)
```
This provides the more verbose output
```
[E] Failed to decode mqtt message: expected set() but found {24, 165}
[*] msg.payload: b'MA\x91\x02\x05\x01\x02\x05F\xc0\x01\x00\x19\x98Be2542fc34-xxxx-1038-8a13-a976d48862fa\x00\x00\x00\x00\xa5\x18\x00\x00\x00\x00\x00\x00\xc3J\x0c\x94\x9c\n`\xec&\xda]\x13h\xcc\xfe\xcbwnb\xb4^I\x00\x9a\xcbJ\xbe\xab\x82\xb7\xf8;\xb3{\xf5\xd0\xb2=\x93\xea\xe2V\xb6\x8d\xb7\xb4\xd4\xdcJ\x85\xbd\xf4E\xcc\xe8\xbc0i\xb01\xf3y7\xdf~\x17bm\xa4n%\xe4\xf7\x88w\x88\xabp\xeb\x05\xa0\\@1\xc5\x83\xe7\xd0\xe45O\xb0EP\xe4\xcb\x18\xd9qJ\x11h\xd1\x05\xb9\x92B\x99\xfc\x93\xa8\x01B\xa5\xf7\xf6W\xfew\xaa\xba\x027\xe8"\xd2\xdf\x13\xd0-\xb1\x84s]\x95:1\xa0\xdcs\x9b\xc8\x1b\xfb\x89\xf4\xaf\xc6\x9ba\x9a\x1e\xf51-\xaa\x8a\xa3e\xa6(M\x8fi\x1f\xbc\x8f\x82u\xec\xa1\xb1\xf2\x8a\xf4Kq\x16y\xedT\x87\xea\x06JXcz\xc4K!\x0c`\xcad\x90_\xeb;\xbe\x1b\x0f\xb1*\xab"\xdak\x98\xb1\x9c\xae\x022\x01?\xa9D\x06R\x96\x90\x06\xabi\xea_m\x9f\xe5\xc3\xd6\xfcv\xcbk\xe4\x9a[\xd9\xf2\x01\xda\x0bp=\xfaO\xc6\xc3`\xab\\\x0e/E2\x98\xbag\xb1\x02\xd7sT2*mQQ<{\xe5\x18\xbc\xef\x9b\x9f\x16\xb8ad\xe2mA\'\x94\x1d\xa3\xd9\xed+}\xd3\xf9\x9a\x93\x96X\xa6r;\x00\xb7Ud\n\xf9V\x83\xcb\x87X-j,W8\xfb\x83\xee/\xecnnW\x1d\xe3\xb2.\x91\xb8b\xceT\xbd\x1bP\x0ez\x11\xe0B\xfa\x9c\x07!e\x93\xe2\xf6\xa6$\x81\xfa\xd7\xaf\xea-!\xc0\x81+\x05d\xfacG\xe4ck\x1f}\x7f\xdf\xba\x85\xd5\xd2\xa4\x86ukn\xb9^v\x9e\xe7zmJ\xf2\xf4\x1d\x91 s\xb4EKH\xb3\xd3j\xc6R\x00\xf2\x86\xc9\x80\x0f5\t\x8a\xe5\x9f\xe5d~\x1a\xd5\xf3\xd0\x92\xb4\x8d\x19\xfd\xe6\xe3./\x04\x1c\xebA\x9a\xa7q9b\x1c\x15\xcav2\x86\xcb\xea\xc2_r\x8a\x15\x8d\x95\r!\x7fKmM\x86N\x13\xc6\xf1a3]*\x94\xabo\xd6\xb0*\xff\xfa\xbe\xb3J\x06]]\xaaz\xae\\\xe6p\xf0\xe5$\xfb\x08\x1b\xd1\x95O\x8f\x14t\xd9\xb5}.J\tG\xb8\xf2\xaf\xfa\xb6\xc6:\x05MT22\xff\x19\xd2\xc1\x1fl\xe1R\x93\xcc\x0e\xc8*D\xa3\xa4\xd7\xd2\xfa\x02aF\xe6\x11G\xcf\xdc,\xd6:\xb7\xe7\xccX\x0b}>>`*\x03\xfa}\xd0\x17\xe3>\xc4>\xd5\x08\xd5\x1a\xbe\x99<\xab\xdf\xc6\xbe?(W\xb6(;\rz\xe2'
[E] Failed to decode mqtt message: expected set() but found {1, 98}
[*] msg.payload: b'MA\xd1\x02\x05\x01\x02\x05F\xc0\x01\x00\x1c\x98Be2542fc34-xxxx-1038-8a13-a976d48862fa\x00\x00\x00\x00b\x01\x00\x00\x00\x00\x00\x00\xc3J\x0c\x94\x9c\n`\xec&\xda]\x13h\xcc\xfe\xcb~\xe8\xb2\x8eq:Vk\x85g9\xaa\xaa\xa9\x93\xdd\x97n\xcf\xe1\x8ev\xab\xf9\xdf\xa6\xed\xa2\xd4\xedC\x98\x1ap\xf2\xec\x16\xdf\xc18\x8b\xa1\xf5\x1c\xa7\xd9q#\x94x\x9b-CU\x01\xb1x\x82\xf6\x92\xc3\x85l-\xbf\x8b8\x9d\xd4\x08Yi\xe9 \n)\x84\xa6`U\r\xfb\xb7\xb6\xf1\x8c\nMS\x81C\xe7~.\xbdI8\xf7T\x01\xc5\n\xae\x91\rr\x08}%\x06\x8aw\x0c\xb2_\x1di\x97\x88\xab7\xd0\xbcS\x8f\xfa\x90\x95\xb0V\x84\xd3\x8a\x86\xdci\xf3\xc6d\xb5-I\x9aK\x15\xe36\x15^\xddr\x937\xfbQ\xceJ\xc5\x8b\xe8(\xe4f2\xe0n5\x9fd\xf3:\x80\xa8\x91aF\x127\xa9e\xf2S\x10\xf1\xdf\x8a\x97\xef\xff\xd0\x08>\x85y\xcb\xda$\x9f\xbbH\xe8\x92"<1\x17\x8e\xe9y\x05M#>Hv\x93\x9e\xb8\x07\xe2\x1f?\xbe\xd49\x9f/\xec\x10j/\xb7%\xde\x84\xc9\'\x0b\xc5f\xa6\x98\xf6$\xa0~\xc1w\n\xf1x\xcc)|\xebC\x98\x1e\xa2*\x9a\xae\r\xb3\xea\xd5\x1d\xc6\xcf\xf0\x82\xfa\xa8\xa0\xf3\xa4\x19\xdc]W|o\xc6 ?\xc9l\xf0\x9fz~\xbd\x17k\xae*\x7f2h\x1e\x9d\x11\x0225\xbe\xbd\x1f\x0c[\xbc\xa1s\xba\xa5\x04\x13|\x0e\xa7\xadc3\x1e\x05\xd8<\xc2\xe6\x87\x07=\x80\x91Y>W\xdbH\x86\xd4\x13\x90\x19\x95\x96\x868\x06g\xbdQ\x0f\x84\x0fM\xf7\xa82\xd0\x88\xaax\xb8\xba`\x17\x1f\x82C;\x7f\xdc7r\xa0\x8f\nX\x18lp\x1f\xb2\xc1\xa74W\xfb\x07\xdbW_\x0bQ\xd0U\x9bY\x15\x88\x1c\x1d\x82Jn\xe47\xc0\t\x9c\x14\xffi![\x03\xe2_\xd8&\xd4\x1bT\xf7\xdf1e\x18\xb5(sH \x889\x8a\x9bShM\xf0\x16\xb5\xc0\xc8\x1a\x8b\xfe\x1e,\x9c+\x8d\xdc\xb9\xb1&O\x0c\xd0\xbc\xd7\x8d7T\x8f\xf6v\xc9L\xab\xd00=\xe4\xebj\xdbo]d\xc5\xfd%\xa7^\xe7--\x12\x1a\x8d\xe2\x84\xf5T>\xd9\xc1w\t,I\x8f\xfd\xb5{\xa0\x8a\x9b;\x8cVDG\x0e\xdf\xf0\xd2$y\x92\xd3\xab/X\x90\xe9\xc5O\x84\x07\t\xa3\x99\x19-\r\n\xa9\t\xa9M\xb3\xdc\xd3T\x9e\xdc/\x0f\xed\x1d\xce\x9c\x83#\xaeK\x85\xde)5\xbc\xdf=J\x07\xea\x84\xee\x95\xa9\x02W;\xae\xc2f-<\xaf\x88V>\x16\x12\xa7\xf3J\x9f\x9b\xee\x8ao\xfc"\xf2\x96\x89I,\xa8\xfe\xc4P\xdf\xa0\x83y\x83\xc6\xf2\x90)\xfdG\x86\xf6M \xb0R\xb2~\x88\xd4\xea\xdb\x9e\xf6P_\xd7\xa9\xf7\t\xc4QQ'
```

billyjbryant, 2023-11-03:
It looks like there is indeed a bug in `ankerctl` with regards to the latest firmware versions of the AnkerMake M5. We will need to investigate this further.

billyjbryant, 2023-11-03:
@Maddin4000 can you please delete https://github.com/Ankermgmt/ankermake-m5-protocol/issues/131#issuecomment-1783294390, it does not contain console logs and is instead causing a lot of interference with reading through the comments on this issue.

billyjbryant, 2023-11-03:
@Maddin4000 please update the title of this issue to `[Bug] MQTT not connecting on latest firmware` 

#132 is a duplicate of this issue.

HansAndreManfredson, 2023-11-10:
I am getting additional error messages:

[image]

I looks like, a language/charset problem? Do you agree?

Hemiu1, 2023-11-11:
Firemware 3.0.70.3.0.84 same problem:

 MqttQueue: Worker started
[E] Failed to decode mqtt message: expected set() but found {217, 19}
[E] Failed to decode mqtt message: expected set() but found {181, 14}

no connection, no camera, no light on/off ... nothing works :-(

mtoner23, 2023-11-27:
I updated my firmware and noticed the same thing.AFter some debuggin it looks like they changed the formatting on mqtt after the guid and perhaps removed the zeroes field. still debugging it locally.

Anyone else looking into this who i can talk with? i tried joining the discord but all the links to it are dead, is there a new discord?

billyjbryant, 2023-11-28:
@mtoner23 you can join the Discord @ https://discord.gg/x3QZr4KcGB

commit bcf68552 (2023-12-04): Changing mqtt padding from zeros to bytes (#134)

closes #131 

Starting with the new firmware releases to anker make printers in the
past few weeks, MQTT no longer connects. There is a field in the mqtt
msg that expects 11 zeroes. these 11 bytes are no longer all zeroes. I
switched the protocol to expect just bytes here since we did nothing
with the zeroes to begin with this should be a backwards compatible
change.

Pasted below is a snippet from my terminal printing out the padding
field. you'll see 2 bytes in the middle of the 11 that are no longer
zeros.

```
[*] Padding: 0x0 0x0 0x0 0x3c  [... cut]
closed 2023-12-04 by just-trey