Ankermgmt/ankermake-m5-protocol #132: [BUG] Web UI stuck in loading

broberts2261, 2023-10-24:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

**Describe the bug**
When I run `python3 ./ankerctl.py webserver run`, I can access the web UI. My login.JSON is properly loaded but the UI is stuck in loading and Im getting the following error in the command line

*] Trying connect to printer 3d Printer (USPRAKM-017051-JPGTV) over pppp using ip <MY_IP> 
[E] Failed to decode mqtt message: expected set() but found {21, 37}

This is continuous

broberts2261, 2023-10-24:
I have tried to troubleshoot this a bit and I think the main issue is the MQTT & PPPP are yellow in the UI and MQTT is not decoding.  

`C:\Users\pc\Desktop\ankermake-m5-protocol\examples>python3 mqtt-connect.py -r us -A 5bb99eb13ce2f2f3480b954eea554a811b72282f6f6ae2f2 -k
[*] Initializing API..
[*] Requesting profile data..
[*] Requesting printer list..
[*] Connecting to mqtt..
[*] Connected to mqtt
ERROR:root:Failed to decode mqtt message: expected set() but found {170, 3}
ERROR:root:Failed to decode mqtt message: expected set() but found {57, 35}
ERROR:root:Failed to decode mqtt message: expected set() but found {35, 21}
ERROR:root:Failed to decode mqtt message: expected set() but found {45, 38}
ERROR:root:Failed to decode mqtt message: expected set() but found {10, 47}
ERROR:root:Failed to decode mqtt message: expected set() but found {25, 76}
ERROR:root:Failed to decode mqtt message: expected set() but found {26, 75}
ERROR:root:Failed to decode mqtt message: expected set() but found {218, 4}
ERROR:root:Failed to decode mqtt message: expected set() but found {250, 13}
ERROR:root:Failed to decode mqtt message: expected set() but found {36, 215}
ERROR:root:Failed to decode mqtt message: expected set() but found {18, 4}
ERROR:root:Failed to decode mqtt message: expected set() but found {3, 206}
ERROR:root:Failed to decode mqtt message: expected set() but found {22, 143}
ERROR:root:Failed to decode mqtt message: expected set() but found {225, 2}
ERROR:root:Failed to decode mqtt message: expected set() but found {65, 19}
ERROR:root:Failed to decode mqtt message: expected set() but found {250, 12}
ERROR:root:Failed to decode mqtt message: expected set() but found {122, 10}
ERROR:root:Failed to decode mqtt message: expected set() but found {8, 87}
ERROR:root:Failed to decode mqtt message: expected set() but found {104, 19}
ERROR:root:Failed to decode mqtt message: expected set() but found {27, 14}
ERROR:root:Failed to decode mqtt message: expected set() but found {21, 7}
ERROR:root:Failed to decode mqtt message: expected set() but found {234, 28}
ERROR:root:Failed to decode mqtt message: expected set() but found {67, 3}
ERROR:root:Failed to decode mqtt message: expected set() but found {140, 36}
ERROR:root:Failed to decode mqtt message: expected set() but found {66, 18}
ERROR:root:Failed to decode mqtt message: expected set() but found {13, 175}
ERROR:root:Failed to decode mqtt message: expected set() but found {4, 206}
ERROR:root:Failed to decode mqtt message: expected set() but found {17, 92}
ERROR:root:Failed to decode mqtt message: expected set() but found {13, 174}
ERROR:root:Failed to decode mqtt message: expected set() but found {122, 3}
ERROR:root:Failed to decode mqtt message: expected set() but found {2, 7}
ERROR:root:Failed to decode mqtt message: expected set() but found {25, 77}
ERROR:root:Failed to decode mqtt message: expected set() but found {98, 4}
ERROR:root:Failed to decode mqtt message: expected set() but found {114, 13}
ERROR:root:Failed to decode mqtt message: expected set() but found {20, 103}
ERROR:root:Failed to decode mqtt message: expected set() but found {3, 230}
ERROR:root:Failed to decode mqtt message: expected set() but found {228, 15}
ERROR:root:Failed to decode mqtt message: expected set() but found {20, 54}
ERROR:root:Failed to decode mqtt message: expected set() but found {33, 46}
ERROR:root:Failed to decode mqtt message: expected set() but found {164, 13}
ERROR:root:Failed to decode mqtt message: expected set() but found {28, 239}
ERROR:root:Failed to decode mqtt message: expected set() but found {156, 21}
ERROR:root:Failed to decode mqtt message: expected set() but found {70, 7}
ERROR:root:Failed to decode mqtt message: expected set() but found {16, 69}
ERROR:root:Failed to decode mqtt message: expected set() but found {5, 22}
ERROR:root:Failed to decode mqtt message: expected set() but found {213, 14}
ERROR:root:Failed to decode mqtt message: expected set() but found {167}
ERROR:root:Failed to decode mqtt message: expected set() but found {122, 14}
ERROR:root:Failed to decode mqtt message: expected set() but found {24, 46}
ERROR:root:Failed to decode mqtt message: expected set() but found {29, 214}
ERROR:root:Failed to decode mqtt message: expected set() but found {229}
ERROR:root:Failed to decode mqtt message: expected set() but found {17, 26}
ERROR:root:Failed to decode mqtt message: expected set() but found {90, 10}`

billyjbryant, 2023-10-27:
Can you advise what firmware version you are on with the AnkerMake and which model of printer you have.

broberts2261, 2023-10-28:
Model: M5
Firmware Version: v3.0.57_3.0.74

Thanks for any help with this!

billyjbryant, 2023-11-03:
Please close this issue in favor of #131 which it is a duplicate of.

billyjbryant, 2023-12-04:
This was resolved by #134

chrivers, 2023-12-04:
@billyjbryant good work!

@broberts2261 we have reason to believe the bug should be resolved in the newest version. Can you update your install, and test again?

billyjbryant, 2023-12-13:
@z-master42 at the current moment, the version has not been increased, I will put in a PR to perform a patch release v1.0.1 so that we can roll out a new docker image.