Ankermgmt/ankermake-m5-protocol #137: [BUG] M5C support - failed to decode mqtt message on current HEAD

velvet-jones, 2023-12-07:
**Describe the bug**
Cannot decode MQTT msessages with various payloads using commit bcf6855.

One of several messages that fail to decode:

[*] Trying connect to printer AnkerMake (EUPRAKM-012816-EJPZG) over pppp using ip 
[E] Failed to decode mqtt message
 Exception: Unsupported mqtt message format (expected 2, but found 1)
 Message : 0x4d 0x41 0x89 0x00 0x05 0x01 0x01 0x05 0x46 0xc0 0x01 0x00 0xac 0xf0 0x9e 0x50 0x04 0x0a 0x00 0x00 0x00 0x00 0x00 0x00 0x91 0x19 0x86 0xc8 0x5c 0x38 0x6b 0x3e 0x53 0x8d 0x8b 0xc2 0xec 0xb8 0xb2 0xbb 0x83 0xcb 0xad 0x14 0x78 0x10 0x38 0xef 0xcc 0xfa 0xa1 0xbd 0x1d 0x86 0x2c 0x9a 0x26 0xaf 0x45 0xbf 0xfb 0x59 0x84 0xc5 0x11 0xd4 0x5d 0x56 0xdc 0x99 0xcc 0x43 0xc5 0x28 0x06 0x0a 0xdd 0xe7 0xac 0x85 0x54 0xfb 0x10 0x87 0x64 0xa8 0xae 0x67 0x6c 0x01 0xce 0xf1 0x31 0xd5 0xcb 0x40 0x52 0xb3 0x30 0x78 0x74 0x1e 0x3a 0xc6 0x2c 0xf0 0x19 0x20 0x24 0x1d 0x94 0xbe 0x15 0xc4 0x7e 0x22 0x47 0x97 0x67 0x1e 0x7d 0x42 0x0a 0x9c 0x36 0xd2 0x17 0x97 0xc3 0x02 0xc5 0xa3 0xd9 0x3e 0x3b 0xc3 0x14

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an M5C printer
2. Upgrade to latest firmware
3. Clone ankermake-m5-protocol
4. Start the ankerctl webserver
5. See error

**Expected behavior**
Expected MQTT messages to decode, MQTT/PPPP buttons to turn green and ability to upload gcode files

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Arch x64
 - Browser Firefox
- Firmware version: AnkerMake M5C firmware version V3.1.25

**Additional context**
N/A

treitmayr, 2023-12-17:
In the source code of AnkerMake Studio there is an option to turn on MQTT tracing, see [AnkerStudio/src/libslic3r/AppConfig.cpp#L132-L133](https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/blob/a095f40be22f6804f3911eeacc4364158e8f005c/AnkerStudio/src/libslic3r/AppConfig.cpp#L132-L133). The config entry `mqtt_trace_enable = 0/1` is read from `AppData\Roaming\AnkerMake Studio Profile\AnkerMake Studio.ini`.
As I have not unpacked "my" M5C yet (waiting for Christmas), I was not able to try it out. If this really works, maybe you can try it and share whatever is printed on the console or to a file!?

velvet-jones, 2023-12-21:
I still have some issues building AnkerStudio on Linux, seems one little issue, so I doubt I will be able to get you something before Christmas unfortunately.

jayknott, 2023-12-31:
I have been having the same issue with my M5C. I dug into the code and logs to see if I could figure it out. I found a few things that might help. Overall, the communications looks similar but there are some differences. I laid out the data from my logs to see if it helps. And then some suggestions/findings at the end.

### AnkerMake Studio Log Example

```
[2023-12-30 12:17:40|0x00008db4]
[info][basetype.cpp:11 PrintLog]recvMqttMessageCallBack:
/phone/maker/AK75CU2D28501161/notice:
[
  {"commandType":1003,"currentTemp":2400,"targetTemp":0},
  {"commandType":1004,"currentTemp":2066,"targetTemp":0}
]
```

### Received Data in ankermake_m5_protocol

#### MQTT Topic

`/phone/maker/AK75CU2D28501161/notice`

#### MQTT Payload

Example payloads in different formats for easy decoding.

With prefix:

```
0x4d 0x41 0x89 0x00 0x05 0x01 0x01 0x05
0x46 0xc0 0x01 0x00 0xf1 0x44 0x6e 0x90
0x44 0x12 0x00 0x00 0x00 0x00 0x00 0x00
0x71 0x0f 0xf6 0x9a 0xa1 0x46 0x8c 0x4d
0xaf 0x3b 0x0a 0xe1 0xa6 0x52 0x69 0xf4
0x50 0x01 0x27 0xfe 0xee 0xfd 0xad 0xc3
0x9e 0xb2 0x1f 0xcb 0xcb 0x21 0x55 0x35
0x97 0xa2 0xce 0x0d 0x6a 0x76 0x22 0x13
0x8c 0xd4 0xdf 0x1b 0x84 0x54 0xcf 0x5a
0x36 0xa4 0x9d 0x34 0xf7 0x1c 0xb2 0x76
0xce 0x7d 0x27 0xac 0xc7 0x61 0xd1 0x1e
0x51 0xaf 0xbf 0xeb 0x17 0xfb 0x58 0xa4
0x3e 0x9e 0xa5 0x9d 0x3f 0x7c 0x76 0x72
0x6d 0x75 0x82 0xea 0xf0 0xd7 0xea 0xbd
0xc6 0xa2 0xb1 0xb0 0xe2 0x91 0x49 0x2b
0xef 0xc3 0x66 0xf7 0x63 0xd1 0x16 0xe3
0xa0 0x1c 0x40 0xcf 0xe1 0x64 0xc6 0xaf
0xc9
```

Without prefix:

```
4d 41 89 00 05 01 01 05
46 c0 01 00 f1 44 6e 90
44 12 00 00 00 00 00 00
71 0f f6 9a a1 46 8c 4d
af 3b 0a e1 a6 52 69 f4
50 01 27 fe ee fd ad c3
9e b2 1f cb cb 21 55 35
97 a2 ce 0d 6a 76 22 13
8c d4 df 1b 84 54 cf 5a
36 a4 9d 34 f7 1c b2 76
ce 7d 27 ac c7 61 d1 1e
51 af bf eb 17 fb 58 a4
3e 9e a5 9d 3f 7c 76 72
6d 75 82 ea f0 d7 ea bd
c6 a2 b1 b0 e2 91 49 2b
ef c3 66 f7 63 d1 16 e3
a0 1c 40 cf e1 64 c6 af
c9
```

Indexed bytes to be referenced in converations:

```
0:    1:    2:    3:    4:    5:    6:    7:
4d    41    89    00    05    01    01    05

8:    9:    10:   11:   12:   13:   14:   15:
46    c0    01    00    f1    44    6e    90

16:   17:   18:   19:   20:   21:   22:   23:
44    12    00    00    00    00    00    00

24:   25:   26:   27:   28:   29:   30:   31:
71    0f    f6    9a    a1    46    8c    4d

32:   33:   34:   35:   36:   37:   38:   39:
af    3b    0a    e1    a6    52    69    f4

40:   41:   42:   43:   44:   45:   46:   47:
50    01    27    fe    ee    fd    ad    c3

48:   49:   50:   51:   52:   53:   54:   55:
9e    b2    1f    cb    cb    21    55    35

56:   57:   58:   59:   60:   61:   62:   63:
97    a2    ce    0d    6a    76    22    13

64:   65:   66:   67:   68:   69:   70:   71:
8c    d4    df    1b    84    54    cf    5a

72:   73:   74:   75:   76:   77:   78:   79:
36    a4    9d    34    f7    1c    b2    76

80:   81:   82:   83:   84:   85:   86:   87:
ce    7d    27    ac    c7    61    d1    1e

88:   89:   90:   91:   92:   93:   94:   95:
51    af    bf    eb    17    fb    58    a4

96:   97:   98:   99:   100:  101:  102:  103:
3e    9e    a5    9d    3f    7c    76    72

104:  105:  106:  107:  108:  109:  110:  111:
6d    75    82    ea    f0    d7    ea    bd

112:  113:  114:  115:  116:  117:  118:  119:
c6    a2    b1    b0    e2    91    49    2b

120:  121:  122:  123:  124:  125:  126:  127:
ef    c3    66    f7    63    d1    16    e3

128:  129:  130:  131:  132:  133:  134:  135:
a0    1c    40    cf    e1    64    c6    af

136:
c9
```

### Findings

-   `m5` (byte 6) is `2` instead of `1`.
-   I had difficulty getting `time` (bytes 13-16) to be the correct datetime. I have tried shifting the bytes a few times and couldn't figure it out.
-   The data won't decrypt. It may be that `iv` is not `3DPrintAnkerMake`.

treitmayr, 2024-01-09:
I  finally succeeded in decoding the packets for my M5C, so here are some of my observations (also from looking at the anker_plugin_network.dll):
* The unencrypted header seems to have only 24 bytes (as supposed to 64 used in MqttMsg.parse()). This makes the data length to decrypt a multiple of 16 (which is necessary).
* With the appropriate change in MqttMsg.parse(), I get a proper decoded byte string
  `b'[{"commandType":1003,"currentTemp":2100,"targetTemp":0},{"commandType":1004,"currentTemp":2046,"targetTemp":0}]\x01'`
* Unfortunately, the Python code raises an exception due to some other unexpected differences, which I have not traced to its origin:
  `Exception: expected 0 but found 116`
* Note: In the plugin I encountered a key "f037242f150c38cc3ac8c5f84733125def2a64858d03261f74f34fce7e226041" used for the decryption, but using that one with ankerctl.py only leads to gibberish in the decoded string.

jayknott, 2024-01-10:
That's helpful. I will use that info to see what I can find. I have been modifying the py file to test. So maybe I can figure something out.

treitmayr, 2024-01-10:
Finally managed to get at least `ankerctl.py mqtt monitor` to work properly - see my changes (including some debug messages) in my fork at
https://github.com/treitmayr/ankermake-m5-protocol/tree/playground-m5c

Note that there are already some changes included for generating mqtt packets, but this is fully untested.

jayknott, 2024-01-13:
@treitmayr I was able to get it running on my server as well! I got full progress while printing a Fortnite medallion for my son.

[image]

Xelinor, 2024-01-13:
Great job! I think you two are the first two to get the M5c working with it.

BerndAllgaeu, 2024-01-14:
Sounds good.  Maybe a stupid question: I'm struggling at the first step to import the configuration. I do not have the login.json file. Do I need a other version of AnkerMake Studio?

Thanks Bernd

Xelinor, 2024-01-14:
AnkerMake studio is not AnkerMake Slicer. Different software.

treitmayr, 2024-01-14:
> Great job! I think you two are the first two to get the M5c working with it.

Thanks! Unfortunately I still cannot upload gcode files (at least via Prusa Slicer) - the first file immediately displays as finished, and a second print just waits forever in the slicer's job list. Have not had the time to check why and how this is supposed to work in the first place. @jayknott, did you manage to print directly from your slicer?

jayknott, 2024-01-15:
@treitmayr I haven't tried to progress any. I will probably try tonight. Has been a busy week.

billyjbryant, 2024-01-15:
If either of you have working fixes for M5C, please feel free to PR them.

treitmayr, 2024-01-16:
@billyjbryant I will probably get the time to create a proper PR from my fork at https://github.com/treitmayr/ankermake-m5-protocol/tree/playground-m5c at the next weekend. Then I would also like to find out why sending a print job did not work for me.
This might be the first opportunity for my serial-to-syslog converter (ESP32 based, infos and code to be released soon) to shine when capturing the M5C's serial console output (where you actually get some messages from the gcode upload). Even if it does not bring more insight, it is still cool ;-)

treitmayr, 2024-01-22:
Added PR#145. Please check for the AnkerMake M5C and M5 (if possible).

ErebusBat, 2024-01-25:
Just adding that I have an M5C that I would be able to test / debug with .