Ankermgmt/ankermake-m5-protocol #140: [BUG]? Web Interface: [WinError 10049] The requested address is not valid in its context

BenBlake79, 2024-01-05:
Your issue may already be reported! Please search on the [issue tracker](https://github.com/Ankermgmt/ankermake-m5-protocol/issues) before creating one.

Done

**Describe the bug**
Webserver starts an can be connected to via browser (but not completely)
Interface in browser only shows temperatures and remaining time and progression when printing, but no video (Loading, please wait...)
Jobs can not be sent via PrusaSlicer (hangs at RESOLVING HOST when using localhost, COMPLETE with 127.0.0.1, but nothing happens)

in Python Window:
 * Serving Flask app 'web'
 * Debug mode: off
[*] WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on http://127.0.0.1:4470
[*] Press CTRL+C to quit
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET / HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/vendor/bootstrap-5.3.0-alpha3-dist/css/bootstrap.min.css HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/vendor/jmuxer.min.js HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/ankersrv.css HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/vendor/bootstrap-icons-1.10.5/font/bootstrap-icons.min.css HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/img/setup/prusaslicer-3.png HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:18] "GET /static/img/setup/prusaslicer-4.png HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/setup/prusaslicer-1.png HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/logo.svg HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/load-screen.svg HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/setup/prusaslicer-2.png HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/icon-hotend.svg HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/icon-time-passed.svg HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/icon-bed.svg HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/vendor/bootstrap-icons-1.10.5/font/fonts/bootstrap-icons.woff2?1fa40e8900654d2863d011707b9fb6f2 HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/img/icon-time-remaining.svg HTTP/1.1" 200 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/vendor/bootstrap-5.3.0-alpha3-dist/js/bootstrap.bundle.min.js HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/vendor/cash.min.js HTTP/1.1" 304 -
[*] 127.0.0.1 - - [05/Jan/2024 10:19:19] "GET /static/ankersrv.js HTTP/1.1" 200 -
[*] MqttQueue: Requesting start
[*] VideoQueue: Requesting start
[*] Connecting printer AnkerMake M5 (EUPRAKM-013140-GGWHN) through make-mqtt-eu.ankermake.com
[*] PPPPService: Requesting start
[*] Trying connect to printer AnkerMake M5 (EUPRAKM-013140-GGWHN) over pppp using ip
[E] PPPPService: Failed to start worker: [WinError 10049] Die angeforderte Adresse ist in diesem Kontext ungültig. Retrying in 1 second.
Traceback (most recent call last):
  File "D:\00-3D-Druck\Install\Ankermgmt for PrusaSlicer\web\lib\service.py", line 103, in _attempt_start
    self.worker_start()
  File "D:\00-3D-Druck\Install\Ankermgmt for PrusaSlicer\web\service\pppp.py", line 48, in worker_start
    api.connect_lan_search()
  File "D:\00-3D-Druck\Install\Ankermgmt for PrusaSlicer\libflagship\ppppapi.py", line 265, in connect_lan_search
    self.send(PktLanSearch())
  File "D:\00-3D-Druck\Install\Ankermgmt for PrusaSlicer\libflagship\ppppapi.py", line 367, in send
    self.sock.sendto(resp, addr or self.addr)
OSError: [WinError 10049] Die angeforderte Adresse ist in diesem Kontext ungültig
[repeating from "[*] Trying connect..." every second]

**To Reproduce**
ankerctl was installed according to documentation
"./ankerctl.py webserver run" didn't work in cmd (in File Directory) nor python runtime (./ for current dir?)
"ankerctl.py webserver run" worked in cmd

**What I Tried**
deactivated all firewalls
restarted System

**Expected behavior**
Video is shown in web interface
gcode can be sent to printer
PPPPService connected

**Screenshots**
can be provided if needed

**Desktop (please complete the following information):**
 - OS: W10 22H2 Build 19045.3803
 - Browser : Firefox, Chrome, Brave - current versions
 - Python Version: 3.12.1

**Additional context**
Maybe there is something to be done or a setting to be changed whats not documented?

Thank You very much in advance ;)

billyjbryant, 2024-01-15:
Is any other service accessing the Printer? Such as the AnkerMake phone app or the AnkerMake slicer?

treitmayr, 2024-01-31:
It looks like the IP address of your printer ist not set in ankerctl's configuration. You can confirm that by typing
`python ankerctl.py config show`

As we just discussed this same issue on discord, it might be good if ankerctl had a way of finding or at least requesting that address.

treitmayr, 2024-02-04:
I just opened PR #150 which will add the possibility to set and update the printer's IP address via CLI and web interface. This should make it easy to fix such issues (at least if the printer is indeed connected to the same network as ankerctl).

BenBlake79, 2024-02-04:
Thank You for your advices. I make do with ankermake studio atm and will make new attempts over the next two or so weeks. The printer is not in the same network as the computer as I do not really trust this thing securitywise and put it into a guest-WLAN. But as far as I understood, this shouldn't be the issue as this protocol uses the ankermake-WAN-connection, or isn't it?

treitmayr, 2024-02-04:
My understanding is that the gcode is sent directly to the printer, not via the AnkerMake cloud service. At least that is my experience with a missing printer IP address vs. correct IP address, and it is consistent with the debug messages from ankerctl:
```
[*] Trying connect to printer AnkerMake M5C (EUPRAKM-[XXXXXX]-LNCZN) over pppp using ip 192.168.1.72
[D] TX  --> PktLanSearch()
[D] Started pppp thread
[D] RX <--  PktPunchPkt(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] TX  --> PktClose()
[D] TX  --> PktP2pRdy(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] RX <--  PktP2pRdy(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] TX  --> PktP2pRdyAck(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'), host=Host(afam=2, port=23048, addr='192.168.1.72'))
[D] RX <--  PktP2pRdy(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] TX  --> PktP2pRdyAck(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'), host=Host(afam=2, port=23048, addr='192.168.1.72'))
[*] Established pppp connection
[D] RX <--  PktP2pRdy(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] TX  --> PktP2pRdyAck(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'), host=Host(afam=2, port=23048, addr='192.168.1.72'))
[*] Going to upload 3081792 bytes as 'InsertTest_M3_0.2mm_PLA_M5-CE.gcode'
[*] Requesting file transfer..
[D] RX <--  PktP2pRdy(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'))
[D] TX  --> PktP2pRdyAck(duid=Duid(prefix='EUPRAKM', serial=13657, check='LNCZN'), host=Host(afam=2, port=23048, addr='192.168.1.72'))
[D] TX  --> PktDrw(chan=0, index=0, data=b'XZYH\x98:\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00ed953070-9888-4d')
[D] RX <--  PktDrwAck(chan=0, count=1, acks=[0])
[*] Sending file metadata..
[D] RX <--  PktDrw(chan=0, index=0, data=b'XZYH\x98:$\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
[D] TX  --> PktDrwAck(chan=0, count=1, acks=[0])
[D] TX  --> PktDrw(chan=1, index=0, data=b'\xaa\xbb\x00\x00\x00\x00\x00\x00\\\x00\x00\x000,InsertTest_M3_0.2mm_PLA_M5-CE.gcode,3081792,33064
[...]
```

BenBlake79, 2024-04-17:
I reinstalled completely and put both the printer and my pc into the same network and I've got the same issue. I found #146 and #149 and I also have no IP for the printer in the Ankermake M5 config - where does the Server get the IP from?

BenBlake79, 2024-04-17:
I now used treitmayr's https://github.com/treitmayr/ankermake-m5-protocol/tree/store-printer-ipaddress build and that worked. But I had to connect the printer to the main wlan - the guest wlan for printer and PC didn't work. I blocked Internet-conectivity for the printer and then MQTT didn't work. So ankerctl seems to use a lan-connection (for streaming and transfering files) and an internet-connection for the rest...

I do not like that an IoT-Device has to run in my local lan with open Internet-Connection - this just feels unsafe. Is there anywhere documented which ports ankerctl and the printer is using to block everything else?

And wouldn't it be a better solution to save the settings in an ini-file that is loaded by the server?