Ankermgmt/ankermake-m5-protocol #142: [BUG] pppp is not connecting to the printer

Danf2873, 2024-01-15:
I'm having trouble connecting to PPPP 
it says `trying to connect to printer over pppp using <your ip>`

Danf2873, 2024-01-15:
I'd like a solution where I can edit the code or something to fix it

billyjbryant, 2024-01-15:
The code is in this repository, it is open source. You are more than welcome to edit and contribute back a PR to implement a fix if you find one.

billyjbryant, 2024-01-15:
The upcoming branch has a fix in place for a few issues that were introduced since the release of the M5C and the updated firmware: https://github.com/Ankermgmt/ankermake-m5-protocol/tree/upcoming/v1.1

Danf2873, 2024-01-15:
> The code is in this repository, it is open source. You are more than welcome to edit and contribute back a PR to implement a fix if you find one.

I do not know how to code well

Danf2873, 2024-01-15:
I keep getting this
TOPIC [/phone/maker/------------------/notice]

Danf2873, 2024-01-15:
i went over to the v1.1 branch and im still having problems

Danf2873, 2024-01-15:
> The master branch has a fix in place for a few issues that were introduced since the release of the M5C and the updated firmware

That's the branch I'm on

Danf2873, 2024-01-15:
[image]

billyjbryant, 2024-01-15:
Do you also have the AnkerMake app or Slicer open? If any other service or app is utilizing the PPPP stream, it won't load on ankerctl. 

Can you look into the console logs of your browser or the server's logs and see what errors are being shown for the PPPP connection?

billyjbryant, 2024-01-15:
> I do not know how to code well

Then why did you ask for a "solution where I can edit the code or something to fix it" ?

billyjbryant, 2024-01-15:
Also ensure that both the ankerctl server and your printer are on the same network and can reach each other. You could test this by pinging your printer's IP from the machine that is running ankerctl, if you are using the docker container you'd need to enter the container in interactive mode to do this.

Danf2873, 2024-01-16:
> Do you also have the AnkerMake app or Slicer open? If any other service or app is utilizing the PPPP stream, it won't load on ankerctl.
> 
> Can you look into the console logs of your browser or the server's logs and see what errors are being shown for the PPPP connection?

`Trying connect to printer Dannys 3d printer (----------------) over pppp using ip -----------`

billyjbryant, 2024-01-16:
Is that the only log line? There are no other errors? Can you run ankerctl with the `--verbose` flag?

scottbilas, 2024-02-03:
I am seeing this same issue. In the dashboard MQTT and CTRL are green, PPPP is yellow. I can see the message stream with updates on temperature etc. while a print is running, but I can't send gcode because of PPPP.

I'm using the `master` branch, both machines are on the same subnet (also M5 pings fine).

Here's a section of the verbose log that is repeating over and over:

```
[D] FileTransferService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
[*] Trying connect to printer AnkerMake M5 (EUPRAKM-002312-CFFHF) over pppp using ip 192.168.5.173
[D] TX  --> PktLanSearch()
[*] TOPIC [/phone/maker/AK7ZRM0C43500559/notice]
[D] 4d41c1000501020546c00100294abe6566376366616333392d366266392d313033622d616231352d636633396630326536316662000000008f2300000000000025b828e4fbaace6b92a93acd2cb54542c3abbfa00911fb63bfeb507766a58a959b51314e43e0603c6cab903aec24e50eee45fddc860fbe81a94f025ab6ac096deec44ae63bc1a29485984d7058c70541e18846203df626ee7fcee4fe15879d6544da545792f4ee80131e0ec64ca7f2099335bf3e6da8d5d5a45f0b1c43d6eb206b
[D] MQTT message: {'commandType': 1003, 'currentTemp': 14900, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 4239, 'targetTemp': 0}
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] FileTransferService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
```

scottbilas, 2024-02-03:
Btw I can debug Python no problem, if you gave me a few hints on where to look. As far as I can see, a PktLanSearch is sent through the socket, but nothing happens after that. It just keeps trying connect over and over.

scottbilas, 2024-02-03:
Packet capture shows me that ankerctl.py is sending `f1300000` over UDP to the M5, which is responding with `f141001445555052414b4d00000009084346464846000000`.

martinmcu, 2024-02-15:
I am seeing similar behavior. I am starting the server with the command `ankerctl.py --pppp-dump pppp.log -v webserver run
`

Here is the output:
```
[*] Logging all pppp traffic to 'pppp.log'
[*] Trying connect to printer AnkerMake M5 (USPRAKM-005296-YXPKR) over pppp using ip 192.168.1.253
[D] TX  --> PktLanSearch()
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[*] TOPIC [/phone/maker/AK7ZRJ0C42500064/notice]
[D] 4d41b1000501020546c0010084bdcd6532353432666333342d336465322d313033382d386131332d61393736643438383632666100000000bc26000000000000c34a0c949c0a60ec26da5d1368ccfecb0eccac5da61db46d01316894ddf245df911a79ca17c7ceff04313ba9b55dd44112cd9b78578818bced5d2680769d231fe2fc4cc40d8e438addcc04a7b03f89a96886175b6a23aa27c9bd5615eea153b5d56e395e9639b3cbfba47743c8e7db0143
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2110, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2223, 'targetTemp': 0}
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
[*] Logging all pppp traffic to 'pppp.log'
[*] Trying connect to printer AnkerMake M5 (USPRAKM-005296-YXPKR) over pppp using ip 192.168.1.253
[D] TX  --> PktLanSearch()
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[*] TOPIC [/phone/maker/AK7ZRJ0C42500064/notice]
[D] 4d41b1000501020546c0010087bdcd6532353432666333342d336465322d313033382d386131332d61393736643438383632666100000000711b000000000000c34a0c949c0a60ec26da5d1368ccfecb0eccac5da61db46d01316894ddf245df963a6006e378062069cc1d87d002273e5ec5ed423e1f7119ce0d0b21d96ccbfab03670409c5f9f45145d5daec9124c32bb74f940bdc48edbbff3c2003045395beb27b9c3474beb4113c62fbe4ed9263ba7
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2105, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2207, 'targetTemp': 0}
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
```
Note that the long hex string is different between the two lines. In the pppp.log file, I just see this
```
[]$ tail pppp.log 
# ========== 2024-02-15T00:30:05.875806 Logging starts ==========
2024-02-15T00:30:05.875951 TX 192.168.1.253:32108 f1300000
# ========== 2024-02-15T00:30:08.879764 Logging starts ==========
2024-02-15T00:30:08.879931 TX 192.168.1.253:32108 f1300000
# ========== 2024-02-15T00:30:11.883207 Logging starts ==========
2024-02-15T00:30:11.883319 TX 192.168.1.253:32108 f1300000
# ========== 2024-02-15T00:30:14.887341 Logging starts ==========
2024-02-15T00:30:14.887451 TX 192.168.1.253:32108 f1300000
# ========== 2024-02-15T00:30:17.891737 Logging starts ==========
2024-02-15T00:30:17.891920 TX 192.168.1.253:32108 f1300000
```
I am on commit `0007f4a`

treitmayr, 2024-02-15:
This should just be a reporting issue, printing should still work with the M5C.
You can try the ankerctl version from my integration branch at https://github.com/treitmayr/ankermake-m5-protocol/tree/m5c-integration which includes all my PRs #145, #148, and #150. This should show a correct PPPP badge and remove the rotating video animation.

martinmcu, 2024-02-15:
Using commit `a3f06e6` of the integration branch [https://github.com/treitmayr/ankermake-m5-protocol/tree/m5c-integration](url), I still see the same result:
```
[*] Logging all pppp traffic to 'pppp.log'
[*] Trying connect to printer AnkerMake M5 (USPRAKM-005296-YXPKR) over pppp using ip 192.168.1.253
[D] TX  --> PktLanSearch()
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[*] TOPIC [/phone/maker/AK7ZRJ0C42500064/notice]
[D] 4d41b1000501020546c0010000cacd6532353432666333342d336465322d313033382d386131332d61393736643438383632666100000000ab10000000000000c34a0c949c0a60ec26da5d1368ccfecb0eccac5da61db46d01316894ddf245dfbe1603516fa3a7a8b902aaa92ddc2768cb8e96ee9a659ce42b64ef9f7e32d01912a142378a7a67ae93f0f048bba038254970f799cbe20eb280f0f165694e983b2b0988bddeff70dfe7ee021c26a9d466b4
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2082, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2203, 'targetTemp': 0}
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
[*] Logging all pppp traffic to 'pppp.log'
[*] Trying connect to printer AnkerMake M5 (USPRAKM-005296-YXPKR) over pppp using ip 192.168.1.253
[D] TX  --> PktLanSearch()
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[*] TOPIC [/phone/maker/AK7ZRJ0C42500064/notice]
[D] 4d41b1000501020546c0010003cacd6532353432666333342d336465322d313033382d386131332d61393736643438383632666100000000490e000000000000c34a0c949c0a60ec26da5d1368ccfecb0eccac5da61db46d01316894ddf245dfd0bf4f9f8336604a81863cc9a031eb855cf60197dc52a728fa5916c8a4291b146d2031569cd6a2838c121e07b51a2fe49868f65d0d524b8a14a64b42bb70a5c624d4607a43c1d2e8c97835dd6c4f30c80c
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2087, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2189, 'targetTemp': 0}
[D] VideoQueue: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
```
The `pppp.log` file is the same
```
# ========== 2024-02-15T01:23:29.091843 Logging starts ==========
2024-02-15T01:23:29.092071 TX 192.168.1.253:32108 f1300000
# ========== 2024-02-15T01:23:32.095460 Logging starts ==========
2024-02-15T01:23:32.095576 TX 192.168.1.253:32108 f1300000
```
Finally, the command `./ankerctl.py -v --pppp-dump pppp.log pppp print-file temptower.gcode` hangs with the following output and a 0-byte `pppp.log`:
```
matt@superlappy: ~/AnkerMake/pppp
[] (m5c-integration %=)$ ./ankerctl.py -v --pppp-dump pppp.log pppp print-file temptower.gcode
[D] Using printer [0]
[*] Logging all pppp traffic to 'pppp.log'
[*] Trying connect to printer AnkerMake M5 (USPRAKM-005296-YXPKR) over pppp using ip 192.168.1.253
[D] TX  --> PktLanSearch()
[D] Started pppp thread
```

treitmayr, 2024-02-15:
@martinmcu Sorry, I was obviously on the wrong track! Hopefully someone else can shed light on what happens in your case.

rcourtman, 2024-02-17:
I think I'm having the same issue here as well. I have an M5C and I'm using @treitmayr ankermake-m5-protocol-m5c-integration branch. I load up the web server, I log in with my details and it successfully pulls in my config just fine. It stores my IP which I can ping just fine. I have green on MQTT and HTTP but the PPPP is yellow. 

 * Running on http://127.0.0.1:4470
[*] Press CTRL+C to quit
[*] MqttQueue: Requesting start
[*] Connecting printer AnkerMake M5C (EU_redacted) through make-mqtt-eu.ankermake.com
[*] Connected to mqtt
[*] MqttQueue: Worker started
[*] TOPIC [/phone/maker/AK_redacted/notice]
[*] PPPPService: Requesting start
[*] Trying connect to printer AnkerMake M5C (EU_redacted) over pppp using ip 192.168.0.187
[*] TOPIC [/phone/maker/A_redacted/notice]

C:\Users\court\Downloads\ankermake-m5-protocol-m5c-integration (1)\ankermake-m5-protocol-m5c-integration>python ankerctl.py pppp lan-search
[*] Printer [EU_redacted] is online at 192.168.0.187

in my config this printer ip and p2p_duid matches what is shown with pppp lan-search

Some extra logs with verbose:

[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[*] TOPIC [/phone/maker/AK_redacted/notice]
[D] 4d418900050_redacted
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2100, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1003, 'currentTemp': 2100, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2053, 'targetTemp': 0}
[D] MQTT message: {'commandType': 1004, 'currentTemp': 2053, 'targetTemp': 0}
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService worker starting..
[*] Trying connect to printer AnkerMake M5C (EU_redacted) over pppp using ip 192.168.0.187
[D] TX  --> PktLanSearch()
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)
[D] PPPPService: Awaiting ready (RunState.Starting)

treitmayr, 2024-02-17:
@rcourtman Could you try with the Windows Defender Firewall disabled to rule out that the packets are dropped by the firewall? If true, this might be the same or similar issue as #49.

I tried this myself now, and initially disabling the firewall allowed the pppp connection to be established. Eventually I ended up adding two firewall rules (one incoming, one outgoing) which allow UDP traffic on all ports for the `python3.12.exe` (not `python.exe`!) executable.

rcourtman, 2024-02-17:
> @rcourtman Could you try with the Windows Defender Firewall disabled to rule out that the packets are dropped by the firewall? If true, this might be the same or similar issue as #49.
>
> I tried this [...]

Boom! worked straight away! Thank you! will add the rules.

scottbilas, 2024-02-24:
Ugh, can't believe I didn't think to check the firewall.

Disabling Defender solved the problem. 🤘🏽 Thank you! Also thank you for the research and building out this tooling. I'm so happy to use Orca to print direct to my M5.

TheCodingCanal, 2024-03-10:
also had the same issue with the same resolution. I added python rules for private network and everything works now

ProPablo, 2025-02-07:
A very simple way to make the python rules in the firewall is just to run this command `python -m http.server 8000`. This brings up the firewall check for you to accept. Im running my environment in conda and it just works after that.

@treitmayr Thank you for your fork btw, helped me actually run the login sequence through the website. If you enable issues on your fork I'd like to add an issue with the CLI login sequence on windows. 

Thank you!