Ankermgmt/ankermake-m5-protocol #147: [BUG] Getting failed to decode mqtt message errors

yourjelly, 2024-01-27:
[image]

I'm not sure the cause of this or what it affects, but I am unable to get a feed from the camera either, not sure if related

billyjbryant, 2024-01-29:
What version are you running on?

Deadeyebarb, 2024-01-29:
MQTT doesnt work for me either,

AnkerCTL in HASS Version: Current version: 1.0.0-1 
AnkerMake M5 version: V3.0.78_3.0.88

Deadeyebarb, 2024-01-30:
Ignore my last message, it works locally I think the HASSIO version is on an old release.

billyjbryant, 2024-01-31:
Duplicate of #137

billyjbryant, 2024-01-31:
The latest version should have fixed this issue, can you pull the latest HEAD?

Deadeyebarb, 2024-02-03:
thanks for the response, I assume that something with the ports is wrong in the container. I will need to check, MQTT seems not to work. In Home Assistant.

ThierryBegin, 2024-05-27:
I'm getting theses with docker.

ThierryBegin, 2024-05-27:
fixed it by doing a local docker build image. the remote one must be outdated