Ankermgmt/ankermake-m5-protocol #149: [BUG] PPPPservice: Failed to start worker. M5C

the2armedmen, 2024-01-31:
I am trying to get ankerctl to work to use a different slicer. I think I got it mostly set up, but when I launch in to the webapp the "PPPP" icon is yellow. MQTT and CTRL are both green. 

When I open the python window thing I have this error "PPPPService: Failed to start worker: [WinError 10049] The requested address is not valid in its context. Retrying in 1 second."

I'm not versed in programming really at all, but am willing to take any help.

I am on Windows 11 and am using a M5C.

billyjbryant, 2024-01-31:
Duplicate of #142 #146

the2armedmen, 2024-01-31:
I was able to figure out that on the webserver page there is no IP in the ankermake m5 config window. I am unsure on windows where to find the file to fix the IP of my printer