Ankermgmt/ankermake-m5-protocol #151: [BUG] Is there a way to use this without needing to install the windows AnkerMaker software?

pghpete, 2024-02-15:
I do not have a windows system at my house and the software will not load properly within a Windows VM due to an openGL compatibility issue.

Is there a way to get or create the login credentials necessary to use this software?

treitmayr, 2024-02-15:
There are two options:
* The traditional way with ankerctl: Run "AnkerMake Slicer" (not the newer "AnkerMake Studio") using wine, log in, and then grab the generated encrypted login.json file and import it in ankerctl.
* With PR #148 you can directly log in with ankerctl via the CLI or the web interface.

In addition there is PR #150 which allows you to store the IP address of your printer in the ankerctl configuration file, otherwise it seems you need to manually add it to the configuration file because it is not part of the login data (at least for me it is not when directly logging in via ankerctl). You could either apply these PRs yourself or try my integration branch - which also adds support for the M5C (it you do not need that, it should still work flawlessly): https://github.com/treitmayr/ankermake-m5-protocol/tree/m5c-integration

pghpete, 2024-02-15:
Thank you @treitmayr I will give those options a go now and report back.

treitmayr, 2024-02-15:
You might want to start with option 2, because it should make things easier :-)

pghpete, 2024-02-15:
@treitmayr Thank you!! I greatly appreciate your help with this. For others that may need to do the same thing...

`git clone --branch m5c-integration https://github.com/treitmayr/ankermake-m5-protocol`
`cd ankermake-m5-protocol`
`./ankerctl.py config login US email@address.com "pw"`

...replacing US with your 2 letter country code and adding your own UN/PW.

treitmayr, 2024-02-15:
Good to know it worked for you! Just make sure that your configuration includes the printer's IP address (CLI: `./ankerctl.py config show` and search for "ip:"). If this is empty, try: `./ankerctl.py pppp lan-search --store` (or use the corresponding button in the web interface).

Phoenix-DH, 2024-07-10:
> @treitmayr Thank you!! I greatly appreciate your help with this. For others that may need to do the same thing...
>
> `git clone --branch m5c-integration [...]

Great, I did get it managed to create the default.json file, afterwards I used the command to get the ip, but when I try to import it in the webserver I get that:
Unexpected Error occurred: 'data'

treitmayr, 2024-07-10:
I suspect there is a misunderstanding - the configuration file used by ankerctl.py is in its own file format and contains all the things ankerctl.py has to know whenever it is (re)started. And this is true for both the commandline version and the web. However, for the original ankerctl.py the only way to generate its config file was to import the old ankermake slicer's config file (which is in a different format).
So please just use the forked version from [anselor:exiles](https://github.com/anselor/ankermake-m5-protocol/tree/exiles) (which includes the mentioned PRs) and you do not have to worry about any config file anymore or even have to know about the existence of either of these config files.

Phoenix-DH, 2024-07-10:
Maybe there is a misunderstanding from my side.
I tried now the fork again, also webserver is running and accepts the config.

I have a home assistant instance running with the ofiicial ankerctl add-on. There I tried to import that generated json file and there then the error occurs.

treitmayr, 2024-07-10:
@Phoenix-DH it seems to you running this issue in multiple threads. Also I am confused why you want to switch to the HA integration version of ankerctl.py - according to [this comment](https://github.com/Ankermgmt/ankermake-m5-protocol/issues/130#issuecomment-2220659576) you should be able to use the exiles version. But I did not try any HA integration myself, so I cannot confirm this. If it is possible, just initialize anterctl.py via the web interface (login + printer search, all right there), and you are done.
As written before, due to my lack of experience with the HA integration, I will not be able to comment more than done above.

Phoenix-DH, 2024-07-11:
I will use home assistant because I do not want to run an additional instance, because this must always be powered to use the home assistant plugin to get current state. This is the case for me.

Thanks for your support here. I think I found a HA compatible version.