Ankermgmt/ankermake-m5-protocol #156: PPPP not working

revil-O, 2024-03-15:
[*] Trying connect to printer AnkerMake M5 (EUPRAKM-******-*****) over pppp using ip 
[E] PPPPService: Failed to start worker: [Errno 61] Connection refused. Retrying in 1 second.
Traceback (most recent call last):
  File "/Users/osc/ankermake-m5-protocol/web/lib/service.py", line 103, in _attempt_start
    self.worker_start()
  File "/Users/osc/ankermake-m5-protocol/web/service/pppp.py", line 52, in worker_start
    msg = api.recv(timeout=(deadline - datetime.now()).total_seconds())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/osc/ankermake-m5-protocol/libflagship/ppppapi.py", line 351, in recv
    data, self.addr = self.sock.recvfrom(4096)
                      ^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionRefusedError: [Errno 61] Connection refused

[image]

OS: MacOSX Sonoma 14.2.1
Python: 3.11
Ankermake M5 Fw: 3.2.17_3.0.94

PrusaSlicer is obviously uploading files ... but printing never starts

following information, repeated every 3 seconds , is shown:

[*] Trying connect to printer AnkerMake M5 (EUPRAKM-******-*****) over pppp using ip 
[*] TOPIC [/phone/maker/AK7Z**************/notice]

If you need additional informations, don't hesitate to ask me for.

Regards revil.O

treitmayr, 2024-03-15:
There seem to usually be two possible issues resulting in the above error:
* The printer IP address is not configured. @revil-O it seems that this is your problem because the message "Trying connect to printer ... over pppp using ip" is missing the IP address. I fixed this by allowing to store a detected printer IP address and warning you if the IP address is not set yet, see PR #150. But you could also try https://github.com/treitmayr/ankermake-m5-protocol/tree/m5c-integration which integrates my three recent PRs.
  or
* The hosts firewall does not allow network traffic to the printer. This was mostly observed for users running ankerctl on an MS Windows machine. There are already various tickets created due to this issue, but ankerctl itself cannot do anything about it. You have to reconfigure your firewall.

anselor, 2024-03-15:
I've attempted to integrate all of the recent pending PRs into a complete set of all known fixes/improvements here:
https://github.com/anselor/ankermake-m5-protocol/tree/exiles

I've built a [docker container](https://cdn.discordapp.com/attachments/1109478707787218964/1218063550770053140/ankerctl_exile.tar.gz?ex=66064d15&is=65f3d815&hm=a8f2ee1bdfb231d53c97a21a16633abd1180aa741e531a2dc22f9fdfb5dbdd73) for it if you want to try it out.

revil-O, 2024-03-16:
hi guys,

i fixed the issue by entering the ip address of my printer manually to default.json.

In Mac OS X the default.json config file is found @ $HOME/Library/Application Support/ankerctl/default.json

btw.
- using pppp lan-search with the store - option gains no success.
- the following mentioned new card is not shown in the web guide

[https://github.com/Ankermgmt/ankermake-m5-protocol/pull/150#issue-2117160112](url)

now everything's works like a charm !!!

Regards revilO

ngiaimo, 2024-03-18:
Hey @treitmayr - Thanks! dopey me didn't notice my firewall got turned back on. I did turn that off and it finally started working but now I am seeing the following:
[image]

pppp keeps failing on and off. Will continue to check a few things on my end.