Ankermgmt/ankermake-m5-protocol #161: Support login for ankermake studio users

rpiterman, 2024-06-17:
Currently, Ankermake offers only the Ankermake Studio.

I use this on mac and the login.json is not used anymore, so no way for me to use ankermake-m5-protocol.

--

Please document / provide login support also for Ankermake studio users.

treitmayr, 2024-06-17:
I recommend you use the [anselor:exiles](https://github.com/anselor/ankermake-m5-protocol/tree/exiles) fork of this repository, which includes my PR #148 for supporting direct login from ankerctl's web interface. Also see this screenshot https://github.com/Ankermgmt/ankermake-m5-protocol/pull/148#issuecomment-1925552423.

Phoenix-DH, 2024-07-10:
I just sae that bug because I think I am hitting exactly by the same issue:
https://github.com/Ankermgmt/ankermake-m5-protocol/issues/163

I did not try this until now, bu as mentioned in my bug I am using the home assistant add-on amd there you must upload a login.json

How can I gez thus managef?

Phoenix-DH, 2024-07-10:
@treitmayr 
I just tried the fork, but I still get the same errors when trying to run import or start the webserver:

> PhoenixBrainBook-3:ankermake-m5-protocol-exiles Phoenix$ ./ankerctl.py config import
> [!] Failed to import file - check if you are logged into Ankerslicer
> PhoenixBrainBook-3:ankermake-m5-protocol-exiles Phoenix$ ./ankerctl.py webserver run
> [W] No printers found in config. Please upload configuration using the webserver or 'ankerctl.py config import'
> [!] Printer number 0 out of range, max printer number is -1 
> [*] ServiceManager: Shutdown complete

treitmayr, 2024-07-10:
@Phoenix-DH if you want to use the console for logging in, just try `./ankerctl.py config login`. This should ask you for the login credentials and your country (I might have changed it from US/EU to a specific country code, cannot remember right now) and the login process will start and generate the needed config file containing the required user-specific key.

treitmayr, 2024-07-10:
After that you may still need to detect and store your printer's IP address in ankerctl's config file. Turn on the printer and wait until it connects to your Wifi, then run `./ankerctl.py pppp lan-search --store`, see my PR #150.

Phoenix-DH, 2024-07-10:
@treitmayr 
Thanks, I did, the file was now with IP in, but the error is still the same on the IP:4470 page.