Ankermgmt/ankermake-m5-protocol #164: [BUG] Printer shows wrong ip when i import the login.json

xkoldxx, 2024-07-18:
As the title says when i import the login.json the printer shows an ip that is not linked to the printer - not sure if this is cached somewhere

boctok, 2024-07-25:
Mine never shows an IP. It is always blank. Also, for some reason, whenever I try to run `./ankerctl.py config show`, a second terminal window pos up and then immediately closes, so I don't ever see any of the configuration info. I can run the webserver, but pppp and mqtt are always yellow. Maybe I am missing a step somewhere in the installation, but I've gone over it at least a dozen times. Clone repo, open cloned directory, start an admin terminal, run the pip command, run ./ankerctl.py, import config, which always seems to go okay. Try to see config, but pop-up terminal window closes. Run web server, but MQTT and pppp are always yellow. Running Python 3.12.4.

treitmayr, 2024-07-25:
Have you tried this?
https://github.com/Ankermgmt/ankermake-m5-protocol/issues/161#issuecomment-2174304748