Ankermgmt/ankermake-m5-protocol #167: [BUG] `config import` fails because `AnkerMake` folder and "login.json" do not exist

heyadamhey, 2024-11-03:
I cloned the repo and installed via CMD. The installation completed but I was not able to import the `login.json` file
The `python3 ankerctl.py config import` command returns the following error:

**Error Message**
`[!] Failed to import file - check if you are logged into Ankerslicer`

AnkerMake Studio is installed and I am logged in
[image]

I checked the AppData and Roaming folders for the path described in the readme, but my installation has no "AnkerMake" folder under Local. There is `AnkerMake Studio` and `Ankermake Studio Profile`
[image]

I searched for "*.json" in both `AnkerMake...` folders with `Local` and `Roaming` and there is no `login.json`. 

**Desktop:**
 - Win 11 Home
 - AnkerMake Studio 1.5.24

tobias-richter, 2024-11-04:
I think you need the first generation of the anker slider software and not the studio in order to get the credentials.

treitmayr, 2024-11-04:
I recommend you use the [anselor:exiles-1.1](https://github.com/anselor/ankermake-m5-protocol/tree/exiles-1.1) fork of this repository, which includes my PR #148 for supporting direct login from ankerctl's web interface or CLI (i.e. no more importing of a login file). Also see my comments to issue #161.

sshanky, 2024-12-30:
I tried today and got this when I tried logging in:

```
[*] Login successful, importing configuration from server..
[*] Initializing API..
[*] Requesting profile data..
[*] Requesting printer list..
[*] Requesting pppp keys..
[!] Config import failed: 'NoneType' object is not iterable
```

pghpete, 2025-10-08:
@treitmayr Your fork is no longer working for me. I get a "TypeError: Arguement of type 'NonType' is not iterable"  when running "./ankerctl.py config login"... not sure if here is the best place to relay any troubleshooting information on it but I didn't see an "issues" page under your fork.  I am desperately trying to use ankerctl so i can remotely print again.