Ankermgmt/ankermake-m5-protocol #171: [BUG] Using exiles in Podman, PPPP won't connect

ChrisMcZork, 2024-12-28:
**Describe the bug**
I'm trying to set up ankerctl in a podman container on openSUSE microOS. MQTT and CTRL are green, PPPP stays yellow.

**To Reproduce**
Set up the default.json like this, IP of the printer was manually entered because autodetect failed as well
"printers": [
    {
      "id": xxxx,
      "sn": "xxxx",
      "name": "AnkerMake M5C",
      "model": "V8110",
      "create_time": 1692950766.0,
      "update_time": 1734777448.0,
      "wifi_mac": "xxxx",
      "ip_addr": "192.168.0.169",
      "mqtt_key": "xxxx",
      "api_hosts": [
        "18.157.241.124",
        "18.168.14.146",
        "15.188.180.60"
      ],
      "p2p_hosts": [
        "18.157.241.124",
        "p2p-mk-lon.eufylife.com",
        "p2p-mk-par.eufylife.com"
      ],
      "p2p_duid": "xxxx",
      "p2p_key": "xxxx",
      "__type__": "Printer"
    }
  ],

Run the container like this:
podman run \
--name ankerctl \
-v /var/container/ankerctl:/root/.config/ankerctl:z \
-p 4470:4470/tcp \
-p 4470:4470/udp \
-e FLASK_HOST=0.0.0.0 \
-e FLASK_PORT=4470 \
anselor/ankerctl:exile-latest

Connect via webinterface, MQTT and CTRL are green, PPPP stays yellow.

Same behaviour if I try to run the container in root-mode.
Firewall (as far as I'm able to tell) is not configured, other containers on the same system can communicate through TCP and UDP incoming.

ChrisMcZork, 2025-01-21:
A short update, tried this again .....

Podman as above, PPPP stays yellow
Running the exact same configurtion in a python virtual environment does work

Maybe @treitmayr or @anselor are there to help?

havedill, 2025-03-04:
Does your podman host have its firewall enabled? Try shutting it off and see if PPPP works after that.

jtthejet, 2025-03-07:
I am running ankerctl on mac I was able to run the program successfully the PPPP was yellow until I went to the settings and clicked on reset ip address then all were green ran the print video feed popped up and was smooth and has been working since