Ankermgmt/ankermake-m5-protocol #173: How to se Ankerctl in 2025 (including home assistant)

tomjpeg, 2025-09-14:
## Current state of ankerctl
Since this repo is no longer maintained, it has become difficult to keep track of how to use ankerctl today.
This version of ankerctl requires a `login.json` file, which is no longer provided by the current eufymake/ankermake slicer. The old legacy slicer that generated it also can't be downloaded anymore.

👉 Recommendation: Use the fork by @anselor. It allows you to log in directly with your EufyMake credentials without needing a login.json file. https://github.com/anselor/ankermake-m5-protocol/

## Hosting ankerctl the easy way
The simplest way to run ankerctl and start printing is with the Home Assistant add-on from @sondregronas.
https://github.com/sondregronas/ankermgmt-hassio-addons

- Just click the installation button in the readme
- Log in through the web UI
- That’s it - you're ready to print

He also provides a ready-to-use Home Assistant card to control your printer.
https://github.com/sondregronas/ankermake-hass-component

theiereman, 2025-09-22:
@tomjpeg maybe try to open a PR on @anselor fork and see if he accepts it

theiereman, 2025-09-26:
@DynaM1ke you can run the `ankerctl.py config login` command to setup